ros2/rclcpp #1963: Error During Colcon Build

SimeonOA, 2022-07-05:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04 
  - 
- Installation type:  Source
  - 
- Version or commit hash: Rolling
  - 
- DDS implementation:
  - 
- Client library (if applicable): rclcpp
  - 

#### Steps to reproduce issue

```
Build ROS 2 from source in a Docker container on Jetson Nano
```

#### Expected behavior
I am building ROS 2 from source in a Docker container. Colcon build currently fails on rclcpp.

#### Actual behavior
```
--- stderr: rclcpp
In file included from /opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/callback_group.hpp:23:0,
                 from /opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/memory_strategy.hpp:25,
                 from /opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/executor.hpp:37,
                 from /opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/executors.hpp:21,
                 from /opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/parameter_client.hpp:36,
                 from /opt/ros/rolling/src/ros2/rclcpp/rclcpp/src/rclcpp/parameter_client.cpp:15:
/opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/client.hpp: In instantiation of ‘std::optional<std::variant<std::promise<typename ServiceT::Response::SharedPtr>, std::tuple<std::function<void(std::shared_future<typename ServiceT::Response::SharedPtr>)>, std::shared_future<typename ServiceT::Response::SharedPtr>, std::promise<typename ServiceT::Response::SharedPtr> >, std::tuple<std::function<void(std::shared_future<std::pair<typename ServiceT::Request::SharedPtr, typename ServiceT::Response::SharedPtr> >)>, typename ServiceT::Request::SharedPtr, std::shared_future<std::pair<typename ServiceT::Request::SharedPtr, typename ServiceT::Response::SharedPtr> >, std::promise<std::pair<typename ServiceT::Request::SharedPtr, typename ServiceT::Response::SharedPtr> > > > > rclcpp::Client<ServiceT>::get_and_erase_pending_request(int64_t) [with ServiceT = rcl_interfaces::srv::DescribeParameters; typename ServiceT::Response::SharedPtr = std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > >; typename ServiceT::Request::SharedPtr = std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Request_<std::allocator<void> > >; int64_t = long int]’:
/opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/client.hpp:551:38:   required from ‘void rclcpp::Client<ServiceT>::handle_response(std::shared_ptr<rmw_request_id_s>, std::shared_ptr<void>) [with ServiceT = rcl_interfaces::srv::DescribeParameters]’
/opt/ros/rolling/src/ros2/rclcpp/rclcpp/src/rclcpp/parameter_client.cpp:545:1:   required from here
/opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/client.hpp:821:12: error: could not convert ‘value’ from ‘std::variant<std::promise<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > >, std::tuple<std::function<void(std::shared_future<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > >)>, std::shared_future<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > >, std::promise<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > > >, std::tuple<std::function<void(std::shared_future<std::pair<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Request_<std::allocator<void> > >, std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > > >)>, std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Request_<std::allocator<void> > >, std::shared_future<std::pair<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Request_<std::allocator<void> > >, std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > > >, std::promise<std::pair<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Request_<std::allocator<void> > >, std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > > > > >’ to ‘std::optional<std::variant<std::promise<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > >, std::tuple<std::function<void(std::shared_future<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > >)>, std::shared_future<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > >, std::promise<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > > >, std::tuple<std::function<void(std::shared_future<std::pair<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Request_<std::allocator<void> > >, std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > > >)>, std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Request_<std::allocator<void> > >, std::shared_future<std::pair<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Request_<std::allocator<void> > >, std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > > >, std::promise<std::pair<std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Request_<std::allocator<void> > >, std::shared_ptr<rcl_interfaces::srv::DescribeParameters_Response_<std::allocator<void> > > > > > > >’
     return value;
            ^~~~~
/opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/client.hpp: In instantiation of ‘std::optional<std::variant<std::promise<typename ServiceT::Response::SharedPtr>, std::tuple<std::function<void(std::shared_future<typename ServiceT::Response::SharedPtr>)>, std::shared_future<typename ServiceT::Response::SharedPtr>, std::promise<typename ServiceT::Response::SharedPtr> >, std::tuple<std::function<void(std::shared_future<std::pair<typename ServiceT::Request::SharedPtr, typename ServiceT::Response::SharedPtr> >)>, typename ServiceT::Request::SharedPtr, std::shared_future<std::pair<typename ServiceT::Request::SharedPtr, typename ServiceT::Response::SharedPtr> >, std::promise<std::pair<typename ServiceT::Request::SharedPtr, typename ServiceT::Response::SharedPtr> > > > > rclcpp::Client<ServiceT>::get_and_erase_pending_request(int64_t) [with ServiceT = rcl_interfaces::srv::ListParameters; typename ServiceT::Response::SharedPtr = std::shared_ptr<rcl_interfaces::srv::ListParameters_Response_<std::allocator<void> > >; typename ServiceT::Request::SharedPtr = std::shared_ptr<rcl_interfaces::srv::ListParameters_Request_<std::allocator<void> > >; int64_t = long int]’:
/opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/client.hpp:551:38:   required from ‘void rclcpp::Client<ServiceT>::handle_response(std::shared_ptr<rmw_request_id_s>, std::shared_ptr<void>) [with ServiceT = rcl_interfaces::srv::ListParameters]’
/opt/ros/rolling/src/ros2/rclcpp/rclcpp/src/rclcpp/parameter_client.cpp:545:1:   required from here
/opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/client.hpp:821:12: error: could not convert ‘value’ from ‘std::variant<std::promise<std::shared_ptr<rcl_interfaces::srv::ListParameters_Response_<std::allocator<void> > > >, std::tuple<std::function<void(std::shared_future<std::shared_ptr<rcl_interfaces::srv::ListParameters_Response_<std::allocator<void> > > >)>, std::shared_future<std::shared_ptr<rcl_interfaces::srv::ListParameters_Response_<std::allocator<void> > > >, std::promise<std::shared_ptr<rcl_interfaces::srv::ListParameters_Response_<std::allocator<void> > > > >, std::tuple<std::function<void(std::shared_future<std::pair<std::shared_ptr<rcl_interfaces::srv::ListParameters_Request_<std::allocator<void> > >, std::shared_ptr<rcl_interfaces::srv::ListParameters_Response_<std::allocator<void> > > > >)>, std::shared_ptr<rcl_interfaces::srv::ListParameters_Request_<std::allocator<void> > >, std::shared_future<std::pair<std::shared_ptr<rcl_interfaces::srv::ListParameters_Request_<std::allocator<void> > >, std::shared_ptr<rcl_interfaces::srv::ListParameters_Response_<std::allocator<void> > > > >, std::promise<std::pair<std::shared_ptr<rcl_interfaces::srv::ListParameters_Request_<std::allocator<void> > >, std::shared_ptr< [... cut ...] ameters_Response_<std::allocator<void> > > > >, std::promise<std::pair<std::shared_ptr<rcl_interfaces::srv::GetParameters_Request_<std::allocator<void> > >, std::shared_ptr<rcl_interfaces::srv::GetParameters_Response_<std::allocator<void> > > > > > > >’
make[2]: *** [CMakeFiles/rclcpp.dir/src/rclcpp/parameter_client.cpp.o] Error 1
make[1]: *** [CMakeFiles/rclcpp.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rclcpp [9min 57s, exited with code 2]

```

#### Additional information

----

fujitatomoya, 2022-07-05:
using docker container ubuntu:20.04 and ubuntu:22.04 with https://github.com/ros2/ros2/commit/28a83a843324918293ccd0d106117cf2fa4e33a6 for source build, i do not have any build problem like this.

SimeonOA, 2022-07-06:
Hi @fujitatomoya 

I am installing ROS2 from source following the instructions here: [from Source](https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html) and [maintaining source checkout](https://docs.ros.org/en/rolling/Installation/Maintaining-a-Source-Checkout.html).

This is my section of code from the Dockerfile: 

```
RUN mkdir -p ${ROS_ROOT}/src && \
    cd ${ROS_ROOT} && \
    wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos && \
    vcs import src < ros2.repos && \
    vcs pull src && \
    
    # Orocos-kdl
    git clone https://github.com/orocos/orocos_kinematics_dynamics && \
    cd ${ROS_ROOT}/orocos_kinematics_dynamics && \
    git submodule update --init && \
   
    
    
    # install dependencies using rosdep
    apt-get update && \
    cd ${ROS_ROOT} && \
    rosdep init && \
    rosdep update && \
    rosdep install -y \
          --ignore-src \
       --from-paths src \
	  --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers ignition-cmake2 ignition-math6 python3-pykdl" && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean && \
    # build it!
    colcon build \
    	--symlink-install \
        --merge-install \
        --cmake-args -DCMAKE_BUILD_TYPE=Release && \   
    # remove build files
    rm -rf ${ROS_ROOT}/src && \
    rm -rf ${ROS_ROOT}/logs && \
    rm -rf ${ROS_ROOT}/build && \
```
    
Please let me know if you need any other information.

alsora, 2022-07-06:
Hi, I had the same problem when cross-compiling ROS 2 Humble from sources for Jetson Nano (aarch64 on Ubuntu 20.04).

The "fix" was to replace that `return value;` with `return std::move(value);`

SimeonOA, 2022-07-06:
Hi @alsora,

Thank you.

I will try this later today.

Were you able to successfully compile the ROS 2 Humble for Jetson Nano with 20.04?  I am trying to do the exact same thing.

Thanks.

smnogar, 2022-07-22:
I hit a similar issue and was able to resolve it with this PR https://github.com/ros2/rclcpp/pull/1976

fujitatomoya, 2022-07-28:
@SimeonOA could you check if https://github.com/ros2/rclcpp/pull/1976 fixes your compile error? (CC: @alsora )

SimeonOA, 2022-07-29:
Hi @fujitatomoya,

@alsora's suggestion fixed the compile error for me. 

I can try #1976 later and see if that works too.

rty813, 2023-08-25:
> I hit a similar issue and was able to resolve it with this PR #1976

this PR works too!

fujitatomoya, 2024-04-04:
closing in favor of https://github.com/ros2/rclcpp/pull/2481

closed 2024-04-04 by fujitatomoya