ros2/rclcpp #1968: shared_futures set by spin_until_future_complete invalid after first get()

fmrico, 2022-07-09:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04
  
- Installation type: binaries

- Version or commit hash: Humble
 
- DDS implementation: Fast DDS

- Client library (if applicable): rclcpp

#### Steps to reproduce issue

In Galaxy, code like this was correct
``` 
  auto request = std::make_shared<plansys2_msgs::srv::GetPlan::Request>();
  request->domain = domain;
  request->problem = problem;

  auto future_result = get_plan_client_->async_send_request(request);

  if (rclcpp::spin_until_future_complete(node_, future_result, std::chrono::seconds(15)) !=
    rclcpp::FutureReturnCode::SUCCESS)
  {
    return {};
  }

  if (future_result.get()->success) {
    return future_result.get()->plan;
```

But in Humble it seems like the data pointed by the returned `shared_future` becomes invalid as soon as `get()` is called for the first time, producing segmentation faults in  subsequent calls to `get()`,  fixing it in this way:

``` 
  auto request = std::make_shared<plansys2_msgs::srv::GetPlan::Request>();
  request->domain = domain;
  request->problem = problem;

  auto future_result = get_plan_client_->async_send_request(request);

  if (rclcpp::spin_until_future_complete(node_, future_result, std::chrono::seconds(15)) !=
    rclcpp::FutureReturnCode::SUCCESS)
  {
    return {};
  }

  auto result = *future_result.get();

  if (result.success) {
    return result.plan;
```

fujitatomoya, 2022-07-12:
@fmrico i am not sure what is the problem here, could you elaborate a bit?

https://github.com/fujitatomoya/ros2_test_prover/blob/master/prover_rclcpp/src/rclcpp_1968.cpp works w/o any error with https://github.com/ros2/ros2/commit/28a83a843324918293ccd0d106117cf2fa4e33a6

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run prover_rclcpp rclcpp_1968
Client: async_send_request call !!!
[INFO] [1657592176.287793495] [client]: Result = 3
[INFO] [1657592176.287849726] [client]: Result = 3
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run prover_rclcpp rclcpp_1968
Client: async_send_request call !!!
[INFO] [1657592184.355723904] [client]: Result = 3
[INFO] [1657592184.355778302] [client]: Result = 3
```

fmrico, 2022-07-12:
Hi @fujitatomoya 

I have just built ros2 from sources (https://raw.githubusercontent.com/ros2/ros2/28a83a843324918293ccd0d106117cf2fa4e33a6/ros2.repos), and I am able to reproduce the bug with your example:

```
fmrico@argo:~/ros/ros2/rolling_ws$ source install/setup.bash
fmrico@argo:~/ros/ros2/rolling_ws$ ros2 run demo_nodes_cpp add_two_ints_server
```

```
fmrico@argo:~/ros/ros2/rolling_ws$ source install/setup.bash
fmrico@argo:~/ros/ros2/rolling_ws$ ros2 run prover_rclcpp rclcpp_1968
Client: async_send_request call !!!
[INFO] [1657607371.285664924] [client]: Result(1st) = 3
terminate called after throwing an instance of 'std::future_error'
  what():  std::future_error: No associated state
[ros2run]: Aborted
```

BTW, I had to modify your test to wait for service, otherwise, I got this error:

```
Client: async_send_request call !!!
cannot publish data, at /home/fmrico/ros/ros2/rolling_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:62 during '__function__'
[ERROR] [1657607708.993082021] [client.rclcpp]: Error in destruction of rcl subscription handle: Failed to delete datareader, at /home/fmrico/ros/ros2/rolling_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/subscription.cpp:52, at /home/fmrico/ros/ros2/rolling_ws/src/ros2/rcl/rcl/src/rcl/subscription.c:183
cannot publish data, at /home/fmrico/ros/ros2/rolling_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:62 during '__function__'
Fail in delete datareader, at /home/fmrico/ros/ros2/rolling_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_service.cpp:103 during '__function__'
[ros2run]: Segmentation fault
```

Do you suggest any other test? Do you need more info about the system?

Thanks
Francisco

fmrico, 2022-07-12:
Hi again @fujitatomoya 

I have debugged a bit, and I have found that the `get()` implies a `std::move`:

```
Thread 1 "rclcpp_1968" hit Breakpoint 1, main (argc=1, argv=0x7ffffffee578) at /home/fmrico/ros/ros2/rolling_ws/src/ros2_test_prover/prover_rclcpp/src/rclcpp_1968.cpp:52
52	  auto a = future_result.get()->sum;
(gdb) s
rclcpp::detail::FutureAndRequestId<std::future<std::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::allocator<void> > > > >::get (this=0x7ffffffed950) at /home/fmrico/ros/ros2/rolling_ws/install/rclcpp/include/rclcpp/rclcpp/client.hpp:81
81	  auto get() {return this->future.get();}
(gdb) s
std::future<std::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::allocator<void> > > >::get (this=0x7ffffffed950) at /usr/include/c++/11/future:802
802	      get()
(gdb) s
804	        typename _Base_type::_Reset __reset(*this);
(gdb) s
std::__basic_future<std::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::allocator<void> > > >::_Reset::_Reset (this=0x7ffffffed820, __fut=...) at /usr/include/c++/11/future:754
754	        explicit _Reset(__basic_future& __fut) noexcept : _M_fut(__fut) { }
(gdb) s
std::future<std::shared_ptr<example_interfaces::srv::AddTwoInts_Response_<std::allocator<void> > > >::get (this=0x7ffffffed950) at /usr/include/c++/11/future:805
805	        return std::move(this->_M_get_result()._M_value());
```
Can it invalidate the next access to the data?

iuhilnehc-ynos, 2022-07-14:
Hi @fmrico ,

I'll create a PR and then ask @ivanpauno to help review it. I am not very clear why [this method](https://github.com/ros2/rclcpp/blob/f43a9198ebb033b43b4bb7b0d8ea2d2798bd566e/rclcpp/include/rclcpp/client.hpp#L607-L616) returns `future` but others([1](https://github.com/ros2/rclcpp/blob/f43a9198ebb033b43b4bb7b0d8ea2d2798bd566e/rclcpp/include/rclcpp/client.hpp#L642), [2](https://github.com/ros2/rclcpp/blob/f43a9198ebb033b43b4bb7b0d8ea2d2798bd566e/rclcpp/include/rclcpp/client.hpp#L673)) returns `shared_future`?

ivanpauno, 2022-07-14:
`async_send_request()` returns a `future`, not a `shared_future`.
To avoid the issue, you need to get the value once and pass a copy or reference to it to all other places, or call `future.share()`.
Though if you're only waiting for the future in `spin_until_future_complete()` and then getting the future, calling `future.share()` doesn't make sense (that makes sense if you're waiting for the future in more than one place, which is not the case).

Note: The `get()` method of `std::future` "extracts" the value from it. The `get()` method of `std::shared_future` returns a reference (or const reference).

ivanpauno, 2022-07-14:
@fmrico if that clears up your doubt, I think we can close this.

fmrico, 2022-07-14:
Hi @ivanpauno 

Yes, now it is clear. 

I would like only to highlight that the problem arises now in (the version packaged in) Humble. In previous versions, it was perfectly possible to call `get()` many times without invaliding the pointer. I suppose many people that use services are now facing this problem migrating their software to Humble. There is no place in tutorials that warns of it. At least, this issue can be useful for many people :smile: 

Thanks!!
Francisco

ivanpauno, 2022-07-14:
> I would like only to highlight that the problem arises now in (the version packaged in) Humble. In previous versions, it was perfectly possible to call get() many times without invaliding the pointer.

Yes, that was a breaking change.

> There is no place in tutorials that warns of it. At least, this issue can be useful for many people smile

I thought that was added to the release notes, but now that I check I don't see a comment related to it.
I don't know if it makes sense to add that now, but it might be a good idea to include it there as it can help people migrate to Humble.

doisyg, 2022-07-19:
Ok, I am having the same issue for my migration from `Galactic` to `Humble`. This is a major breaking change. Most of my async service calls are now broken. This should have been documented somewhere.
Any updated example / best practice on how to do an async service call, wait for some time and then access the result ?

ivanpauno, 2022-07-19:
I've opened https://github.com/ros2/ros2_documentation/pull/2885, let me know if you have feedback about it.

closed 2022-07-20 by ivanpauno