ros2/rclcpp #1980: Create a timer that does not get triggered as soon as it is initialized.

Voldivh, 2022-08-03:
## Feature request

#### Feature description

Create a timer that does not get's triggered at the moment of initialization. There are several ways to create a timer which executes periodically a callback, I would like to have some control over when the timer starts the countdown. At the moment, the workaround is to create the timer, then cancel it and reset it at the moment I would like it to begin, but, in the time between the timer is created and canceled, the callback may be executed.

#### Implementation considerations

A way I thought for this feature is to add another argument to the GenericTimer class such as `start_on_initialization`, this argument can be used as a flag in order to start or not the timer. We could give the argument a default value of true that way, all the implementations so far would't have to be modified and the users could choose the behavior that works best for the application. In order to start the clock in the cases where the argument is false, it is possible to use the reset function and set the flag to false or create a new function `start()` that modifies the flag and starts the clock.

Voldivh, 2022-08-10:
An implementation example that solves this issue can be found in the following forks:
- https://github.com/Voldivh/rcl 
- https://github.com/Voldivh/rclcpp
The solution presented before modifies the `rcl_timer_init` arguments implementing an `autostart` flag. This is later used by the rclcpp API to give access to the user to select if the timer will start on initialization or not.

fujitatomoya, 2023-06-30:
I will go ahead to close this since https://github.com/ros2/rclcpp/pull/2005 has been merged.

closed 2023-06-30 by clalancette