ros2/rclcpp #1983: "yes" and "no" string parameters are confused for bool parameters

Alberto-Neri, 2022-08-09:
- Operating System:
  Windows 10
- Installation type:
  ROS2 Foxy binaries
- DDS implementation:
  Cyclone-DDS
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue
Use whatever node you want and add somewhere:
```
std::string calib;
try {
    this->declare_parameter<std::string>("calib", "no");
    this->get_parameter_or("calib", calib, std::string("no"));
}
catch (std::exception e) {
    std::cout << e.what() << "\n";
}
std::cout << calib << "\n";
```
Then try run the node with: 
```
ros2 run pkg_name exe_name --ros-args -p "calib:=yes"
```
#### Expected behavior
The CLI prints yes.
#### Actual behavior
Exception triggered with description:
`parameter 'calib' has invalid type: expected [string] got [bool].`
#### Additional information
If I launch the node with: 
`ros2 run pkg_name exe_name --ros-args -p "calib:=ok"`
I get the **expected behaviour** (ok printed).

If I modify the code to accept a **bool** parameter, like this:
```
bool calib;
try {
    this->declare_parameter<bool>("calib", false);
    this->get_parameter_or("calib", calib, false);
}
catch (std::exception e) {
    std::cout << e.what() << "\n";
} 
std::cout << calib << "\n";
```
and do: `ros2 run pkg_name exe_name --ros-args -p "calib:=yes"`
I get **no exception and it prints 1**, that is the same as true!!
While doing: `ros2 run pkg_name exe_name --ros-args -p "calib:=ok"`
gives me the inverse exception as before:
`parameter 'calib' has invalid type: expected [bool] got [string].` !!! 

Is this expected?

clalancette, 2022-08-09:
> Is this expected?

Unfortunately, yes, though it is a bug.  We interpret a bunch of this stuff as YAML, and since "yes"/"no" gets automatically converted to a boolean in YAML, that's how it ends up here.

We partially fixed this in https://github.com/ros2/ros2cli/pull/684 , where you can now force the type to be used in `ros2 param` by using the YAML `!!str`, but we didn't fix it in `rclcpp` or elsewhere.

For now, I'd say the best workaround is to avoid `yes` and `no` (and `true` and `false` for that matter) in your strings.  It's not ideal, but until we have full support for the `!!str` syntax, it is the best we can do.

iuhilnehc-ynos, 2022-08-19:
> and since "yes"/"no" gets automatically converted to a boolean in YAML, that's how it ends up here.

I am afraid it's not correct. The "yes"/"no" or 'yes'/'no' can be parsed as a string.
Adding '!!str' seems good to me.

Barry-Xu-2018, 2022-08-19:
>  I am afraid it's not correct. The "yes"/"no" or 'yes'/'no' can be parsed as a string.

Yes. 
Using `"` or  `'` can set `calib` correctly.  
```
ros2 run pkg_name exe_name --ros-args -p 'calib:="yes"'   
```
Using `!!str`  is another way  (https://github.com/ros2/rcl/pull/999)
```
ros2 run pkg_name exe_name --ros-args -p 'calib:=!!str yes' 
```

fujitatomoya, 2023-01-04:
@Alberto-Neri direct solution to this issue would be just adding quote mentioned in https://github.com/ros2/rclcpp/issues/1983#issuecomment-1220433826, we have also added yaml string tag in https://github.com/ros2/rcl/pull/999.

I will go ahead to close this issue, if you have any more questions or issues, feel free to reopen.