ros2/rclcpp #1998: Parameter and parameter event callbacks silently fail if handle is not captured

jasonbeach, 2022-08-18:
## Bug report
In rclcpp, in `rclcpp::ParameterEventHandler` the functions `add_parameter_event_callback` and `add_parameter_callback` return a shared_ptr to a "handle" object. The shared_ptr to the handle is the only owning pointer to the handle, meaning if it's not captured, the handle is deallocated and the callback silently never fires (internally, the `parameter_callbacks_` map holds only weak_ptrs to the callback handles which do get checked when the callback needs to fire. If the handle object no longer exists, it correctly doesn't fire the callback.) 

#### Steps to reproduce issue
In the demos repository, in demo_nodes_cpp-> parameters->parameter_event_handler.cpp, modify this line:
https://github.com/ros2/demos/blob/rolling/demo_nodes_cpp/src/parameters/parameter_event_handler.cpp#L91

from:
`auto handle1 = param_subscriber->add_parameter_callback(param_name, cb1);`

to
`param_subscriber->add_parameter_callback(param_name, cb1);`

run the demo and set the parameter and observe the callback never fires. 

#### Expected behavior
I would expect the callback to fire or if this design really is intentional get some sort of warning of what's happening.

It seems strange to me that the `parameter_callbacks_` map and `event_callbacks_` list hold only weak_ptrs. Only thing I can thing of is that its sort of an implicit way of ensuring whatever object registered the callback still exists when it's time to fire the callback. If this is the case it really should be documented better and at a minimum those functions should be marked `[[no_discard]]` so that if a user doesn't capture the handle they get a compiler warning. 

I think it's perfectly valid to not capture the handle if I have no interest in deleting the callback later and so I think it's fine to change the  weak_ptrs in `parameter_callbacks_` and `event_callbacks_` to shared_ptrs (actually I don't even think pointers need to be used here...just store the callback directly in the map).  I'm trying to work through a scenerio in which you would run into ownership issues....if I come up with one, I'll post it.

clalancette, 2022-08-18:
This is definitely part of the current design that you need to hold onto the pointers.

I'm not intimately familiar with how this is implemented, but most of the places we have weak_ptr usage is because there is some circular dependencies between components, and using weak_ptrs is one way to break it.  Maybe @ivanpauno can shed more light on the current design.

> If this is the case it really should be documented better 

The documentation for it does [say](https://github.com/ros2/rclcpp/blob/3d69031d2a614cf02dc10ea5db153471ea32b1f2/rclcpp/include/rclcpp/node.hpp#L1089-L1090):

```
Resetting or letting the smart pointer go out of scope unregisters the callback.
`remove_on_set_parameters_callback` can also be used.
```

I'm not sure how much clearer it could be, but we are willing to take PRs to improve the wording.

> and at a minimum those functions should be marked [[no_discard]] so that if a user doesn't capture the handle they get a compiler warning.

Ah, I didn't know about that.  It's new with C++17, so we could conceivably add it now.  We'd definitely consider a PR that adds that.

jasonbeach, 2022-08-18:
> The documentation for it does [say](https://github.com/ros2/rclcpp/blob/3d69031d2a614cf02dc10ea5db153471ea32b1f2/rclcpp/include/rclcpp/node.hpp#L1089-L1090):
>
> ```
> Resetting or letting the smart [...]

That's in the node class definition, not in the `ParameterEventHandler`. I had been looking at the demos and then at the `ParameterEventHandler` class itself. I can look at putting a PR together for that and possibly adding `[[no_discard]]`

`add_on_set_parameters_callback` seems a little too clumsy to be useful if I understand it correctly (basically in each handler I would add I would have to iterate through the parameters and then individually call a specific update / validation function (or whatever) for a given parameter.  I think `ParameterEventHandler::add_parameter_callback` is the most useful function, although it doesn't look like you can do much in the way of validation on the param with it. It would be nice to have a similar function on the Node itself so it's not so hidden (it was purely by happenstance that I was looking at the demos and saw the parameter event handler and found out about it).

What I would like to see that I think would even be better would be to add two functions to the Node class: 1) one similar to  `add_parameter_callback` where I can be notified of local or remote parameter changes but assume someone else is declaring and validating them. 2) one called something like `declare_parameter_with_callback` (or even just add another overload of `declare_parameter`) that declares a local parameter and registers the callback (with signature `bool(const rclcpp::Parameter&)`) that can potentially fail if an invalid value is received. Would that be of interest?

fujitatomoya, 2022-08-19:
> Only thing I can thing of is that its sort of an implicit way of ensuring whatever object registered the callback still exists when it's time to fire the callback.

i think that is the case. and probably user application expects that when this object is deleted, callbacks registered by this object will not be called?

> it really should be documented better and at a minimum those functions should be marked [[no_discard]] so that if a user doesn't capture the handle they get a compiler warning.

:+1: this is good idea

jasonbeach, 2022-08-20:
PR is submitted to add the compiler warning if return object is not captured. 

I still think the parameter registration / callback could be a little better--today I made a class that inherited from lifecycle_node and added a ParameterEventHandler class to it.  I then added two functions:

declare_parameter_with_callback(param_name, default_value, callback, parameter_descriptor)
add_parameter_callback(param_name, callback, node_name)

and in my limited playing around with them they worked really nice. The callback in the first one isn't used to validate the parameter, just to do something once it's changed. I don't remember if setting param limits in the parameter_descriptor will do anything. I think adding it to both the `rclcpp::Node` class and rclcpp_lifecycle::Node classes would be useful--just wondering what others think. so far I've only tested them setting parameters in the nodes themselves...I haven't tried with setting the parameters from config file, etc. yet.

fujitatomoya, 2022-08-31:
@jasonbeach i will go ahead close this issue since https://github.com/ros2/rclcpp/pull/2000 has been merged. thank you very much for the contribution :+1: 

> I think adding it to both the rclcpp::Node class and rclcpp_lifecycle::Node classes would be useful--just wondering what others think.

on this, probably opening an another issue would be nice.

closed 2022-08-31 by fujitatomoya