ros2/rclcpp #2007: SEGFAULT in prune_requests_older_than() if pruned_requests are not provided

akela1101, 2022-08-30:
Segfault happens here: 
https://github.com/ros2/rclcpp/blob/dab9f5e0a331b9694c4b5aceb8263d49aa052ecb/rclcpp/include/rclcpp/client.hpp#L772

The documentation says 
```
[inout] pruned_requests Removed requests id will be pushed to the vector if a pointer is provided.
```
So I assume no pointer is a valid input.

Please, fix it, or tell me which branch to submit the fix, so all supported versions were updated.

clalancette, 2022-08-30:
It's hard for me to tell exactly what your problem is from the description.

Nevertheless, if you have a fix that you'd like to see applied, the best thing to do is to create the fix and open it against the `rolling` branch.  Make sure you explain exactly what problem you were having, and how your change fixes it.

Assuming it is reviewed and accepted there, it can be considered for backport to the other stable branches.

akela1101, 2022-08-30:
I updated the description and added a PR. Please, check.

clalancette, 2022-08-30:
> I updated the description and added a PR. Please, check.

Thanks, that is appreciated.

PR #2008 (2022-12-13): fix nullptr dereference in prune_requests_older_than
Fixes #2007