ros2/rclcpp #2020: rclcpp shutdown abnormally when rclcpp inited with rclcpp::SignalHandlerOptions::None

uupks, 2022-10-04:
## Bug report

**Required Info:**

- Operating System:
  - Android NDK r23c
- Installation type:
  - source
- Version or commit hash:
  - humble
- DDS implementation:
  - cyclonedds
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
int main(int argc, char *argv[])
{
    rclcpp::init(argc, argv, rclcpp::InitOptions(), rclcpp::SignalHandlerOptions::None);
    rclcpp::shutdown();
    return 0;
}
```

#### Expected behavior
exit normally

#### Actual behavior
```bash
[DEBUG] [1661888770.858106131] [rcl]: Shutting down ROS client library, for context at address: 0xb4000071f661c7f0
[DEBUG] [1661888770.858455141] [rclcpp]: SignalHandler::uninstall(): notifying deferred signal handler
terminating with uncaught exception of type std::__ndk1::system_error: thread::join failed: Invalid argument
Aborted
```

#### Additional information
[rclcpp](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/src/rclcpp/signal_handler.cpp#L194)

If rclcpp is inited with rclcpp::SignalHandlerOptions::None,  signal_handler_thread will not be created.

----
## Feature request

#### Feature description

#2019 

```c++
if(signal_handler_thread_.joinable()) {
  signal_handler_thread_.join();
}
```

#### Implementation considerations

closed 2022-10-14 by uupks

dawierha, 2024-05-23:
How did you solve the problem?

mjcarroll, 2024-05-23:
This was fixed via: https://github.com/ros2/rclcpp/pull/2019