ros2/rclcpp #2028: Topic cannot be re-subscribed while use_sim_time is true

Barry-Xu-2018, 2022-10-18:
## Bug report

Test is
[image]

While turning on checkbox of `PointCloud2`, you should see point cloud in the right window. 
[image]

If quickly turn on and off checkbox many times, you will find no point cloud is showed, even if checkbox of `PointCloud2` is turned on. 

[image]

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binary
- Version or commit hash:
  - rolling/humble
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
The source code is https://github.com/Barry-Xu-2018/ros2_bug_reproduce_pkg_20221018  

How to run
```bash
$  ros2 launch test_pkg test.launch.xml
```
Quickly turn on and off checkbox of PointCloud2 many times. 

#### Expected behavior

Point cloud is showed correctly while the checkbox of PointCloud2 is turned on.

#### Actual behavior

Sometimes, point cloud isn't showed while the checkbox of PointCloud2 is turned on.

#### Additional information

Only happy while use_sim_time is true.

Barry-Xu-2018, 2022-10-18:
After checking code, find the cause.  
While use_sim_time is set as 'true', it will create a clock thread to run single thread executor for clock subscription (https://github.com/ros2/rclcpp/blob/dec766228dae4e050d37c8d5b8185383fdec76ab/rclcpp/src/rclcpp/time_source.cpp#L383-L389).
And there is another executor in laserscan_to_pointcloud.
Note that the notify condition of the same node is added to guard condition handles of 2 executors.  
If executor in clock thread deals with trigger of guard condition for creating subscription (https://github.com/ros-perception/pointcloud_to_laserscan/blob/272ca4df83ae5faef5df9d4fe6bd90a872360c16/src/laserscan_to_pointcloud_node.cpp#L114), another executor may be always in Executor::wait_for_work() and has no chance to add subscription to be waited on.

Barry-Xu-2018, 2022-10-18:
The root cause is the same as https://github.com/ros2/rclcpp/issues/1611.  
So the PR #1640 (under review) should resolve this problem.

fujitatomoya, 2022-10-26:
@Barry-Xu-2018 https://github.com/ros2/rclcpp/pull/1640 has been merged, can you confirm that `rolling` can solve this problem?

closed 2022-10-27 by Barry-Xu-2018