ros2/rclcpp #2037: MultiThreadedExecutor CPU affinity aware

fujitatomoya, 2022-10-31:
## Feature request

#### Feature description

`MultiThreadedExecutor` should honor CPU affinity to be set by user application for process space, including inherited affinity.
https://github.com/ros2/rclpy/pull/1031 includes this configuration, so the same behavior should go to `rclcpp`.
Using single thread with `MultiThreadedExecutor` will print the warning since it is not recommended, to avoid the possible problem which is described https://github.com/ros2/rclcpp/issues/2029.
But if the CPU affinity is set by user application, it will honor that configuration.

#### Implementation considerations

- Implement abstracted function in `rcutils` so that we can conceal the platform implementation in client libraries.

#### Related PRs

- https://github.com/ros2/rclpy/pull/1031
- https://github.com/ros2/rclcpp/pull/2030

smorita-esol, 2022-11-30:
It would be nice if the new feature included not only the core number but also a configuration set of pinned cores and priorities for each thread.
The existing MultithreadExecutor does not care about which core each thread is assigned to and what priority is set for each.
In a Linux environment, These thread attribute settings could be partially done by using CPU_SET or chrt features, but there might be some environments that do not have such features.

fujitatomoya, 2022-12-01:
> It would be nice if the new feature included not only the core number but also a configuration set of pinned cores and priorities for each thread.

i am not sure about this. probably this is out of scope from this issue atm.

> The existing MultithreadExecutor does not care about which core each thread is assigned to and what priority is set for each.

So is `SingleThreadedExecutor`. For me, this can be applied to `SingleThreadedExecutor`, not only for `MultiThreadedExecutor`.
Even if we can set the thread priority for each thread in `MultiThreadedExecutor`, we cannot control which thread can take which event. I think it makes more sense that if we can set the priority for the event in this `MultiThreadedExecutor` case?

i might be mistaken. if that so, could you elaborate the scenario with `MultiThreadedExecutor`?

smorita-esol, 2022-12-01:
> i am not sure about this. probably this is out of scope from this issue atm. 

Sorry for my misunderstanding. The title includes "Core affinity," so I thought you were also interested in implementing the core affinity settings (pinned core) for each executor thread, not only the number of cores.

>So is SingleThreadedExecutor. For me, this can be applied to SingleThreadedExecutor, not only for MultiThreadedExecutor.
Even if we can set the thread priority for each thread in MultiThreadedExecutor, we cannot control which thread can take which event. I think it makes more sense that if we can set the priority for the event in this MultiThreadedExecutor case?

Almost totally agree. However, I'm not sure if the controlled unit should be "event," "event chain," or other things, so far. 
But, even at this point, it seems beneficial for us to discuss how to control the scheduling parameters (including core affinity) for each thread belonging to the executor thread pool.  

> i might be mistaken. if that so, could you elaborate the scenario with MultiThreadedExecutor?

Thanks for the suggestion, but I have to investigate some more. 
When done, I'll post it in another proper place (ROS Discourse?).

fujitatomoya, 2022-12-01:
sorry for the confusion. thanks for the information, that can be also discussed.

> But, even at this point, it seems beneficial for us to discuss how to control the scheduling parameters (including core affinity) for each thread belonging to the executor thread pool.

that could be useful for mission critical application. i believe those topics have been discussed in micro-ROS community as well.

CC: @JanStaschulat @ralph-lange

ralph-lange, 2022-12-01:
> Even if we can set the thread priority for each thread in `MultiThreadedExecutor`, we cannot control which thread can take which event. I think it makes more sense that if we can set the priority for the event in this `MultiThreadedExecutor` case?

Besides the fact that the current MultiThreadedExecutor interface does not allow assigning certain events (subscriptions, timers, etc.) to specific threads, priority inheritance on the `wait_mutex_` of MultiThreadedExecutor would be required for querying new work from the middleware layer (`rmw`) using the `wait_set`.

This querying works as follows: The first thread of the MultiThreadedExecutor that detects that the current `wait_set` has been taken completely calls `rcl_wait` to populate the wait_set with new events from the middleware layer. When another thread of the MultiThreadedExecutor finishes its current work, it will not call rcl_wait again, but simply wait for the first thread to return from rcl_wait with the updated `wait_set`. If this second thread would have a higher priority than the first thread, the first thread would have to inherit the second thread’s priority to avoid that it is preempted by another thread with medium priority between the first and the second thread.

Note that since Galactic, callback groups of the same node may be distributed to multiple executor instances using `add_callback_group`, i.e., one may create multiple MultiThreadedExecutor and/or SingleThreadedExecutor instances prioritize the thread(s) of each instance individually and distribute the callback groups specifically to them.

JanStaschulat, 2022-12-01:
> It would be nice if the new feature included not only the core number but also a configuration set of pinned cores and priorities for each thread.

In micro-ROS, we have designed a "Dispatching Executor", which works as follows:
configuration step: 
- similarly to adding a subscription to a node in rclcpp, the user adds a subscription to the rclc-executor, but here the user adds also a scheduling policy (e.g. a priority)
- the scheduler creates for each handle (subscription,timer) a dedicated worker thread with the configured scheduling policy. This thread is later responsible for executing the callback function.
at runtime:
- The Executor runs in a dedicated thread and checks for new data at the DDS queue (with rcl_wait)
- If new data is detected, it notifies the corresponding worker thread via a condition variable. Data of the message is not copied, but the pointer to that newly arrived data is made available to corresponding worker thread
- the worker thread wakes up and executes the callback function

See this publication [Budget-based real-time Executor for Micro-ROS](https://arxiv.org/abs/2105.05590), which describes the dispatching executor for NuttX, a real-time operating system which supports reservation-based scheduling (like SCHED_DEADLINE in Linux).  

The[ pull request for the Dispatching Executor on rclc](https://github.com/ros2/rclc/pull/87) will be merged with a new package `rclc_dispatching_executor`. 

In the last meeting of the Real-Time Working Group, I have a short overview presentation about [Real-Time Execution Management in ROS 2](https://docs.google.com/presentation/d/1YkSRiGy256r0AWRs0rdH9Q1_FQDk3B2c/edit#slide=id.p1). Slide 25 shows a diagram of this "Dispatching Executor".

So, the rclc-Executor design is different to rclcpp Executor, that not each thread in the thread pool calls rcl_wait/rcl_take, but only one dedicated thread is responsible for that. The rclc-Executor povides currently a one-to-one correspondence of a worker thread with a (real-time) priority to a particular handle (subscription,timer). This concept could be extended such that one thread is responsible for multiple handles. 

As an alternative, one could also design a real-time Executor in rclcpp using multiple callback groups. Each callback group would be executed in its own executor and own real-time thread. Then the user could choose the callback group, and thereby, configuringt the priority of the thread, in which the callback function is executed. 

Several people in the community have been requesting some real-time configuration in ROS 2. As a first step, we could setup an example with callback-groups and then design a simple API, so that the user only has to specifiy the priority and the organizational stuff (like callback groups, creation of real-time threads is done in the background).

fujitatomoya, 2022-12-01:
@ralph-lange @JanStaschulat 

thank you so much for introducing information and explanation in detail. really informative.

currently `rclcpp` or `rclpy` does not have any capability to control, thread affinity, priority, and matching thread and event at all. it just processes everything with single or multi threads equally. (order to take event is statically implemented too.)
i am not really sure if executor should be capable of doing this, but definitely worth to discuss more in real-time use cases.

@smorita-esol hope this helps you. probably we can open the new issue for this discussion.

I would like to have this issue dedicated to be a counter part of https://github.com/ros2/rclpy/pull/1031. And adjusted the title a bit.

smorita-esol, 2023-02-17:
FYI

> \> i might be mistaken. if that so, could you elaborate the scenario with MultiThreadedExecutor?
> Thanks for the suggestion, but I have to investigate some more.
When done, I'll post it in another proper place (ROS Discourse?).

I've created an issue and implemented a proposal code to control the thread OS scheduling parameters as below.
https://github.com/ros-realtime/ros-realtime.github.io/issues/18

If you are interested in the feature, I'd appreciate it if you could make some comments on it.