ros2/rclcpp #2042: Component Container Masks Names of Internal Nodes of Components (DRAFT ISSUE, WIP)

methylDragon, 2022-11-03:
## Bug report

**Required Info:**

- Operating System:
	- Ubuntu 22.04, Ubuntu 20.04
  - 
- Installation type:
  - Binaries
- Version or commit hash:
  - rolling, 17.1.0
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I made a minimal example!

See: https://github.com/methylDragon/bug-reports-and-pr-examples

Clone and build the repo
```
mkdir -p bug_ws/src
cd bug_ws/src
git clone -b rclcpp/component-container-masking https://github.com/methylDragon/bug-reports-and-pr-examples.git
cd ..

colcon build
source install/setup.bash
```

Now
```
# Terminal 1:
ros2 launch composition composition_demo.launch.py

# Terminal 2:
ros2 node list 
```

The launchfile launches two node components in a component container:
- Talker
- Listener

But in Talker's code, it spins up another internal node named "internal_node".

#### Expected behavior
The nodes listed should be:
```
$ ros2 node list
/launch_ros_3449385
/listener
/internal_node
/my_container
/talker
```

#### Actual behavior
The component container's name masks the internal node's name. This causes a whole host of issues relating to node name resolution (e.g. parameter resolution, etc.)

```
$ ros2 node list
WARNING: Be aware that are nodes in the graph that share an exact name, this can have unintended side effects.
/launch_ros_3449385
/listener
/my_container
/my_container
/talker
```

#### Additional information
I did some digging into this a couple months back, but I sort of forgot.. But the reason this happens is something about the node context/node options gets forgotten/ignored, that causes the internal node's name to get remapped to the component container's.

Also, this issue only occurs when using a component container, a normal manual composition works as expected:
```
# Terminal 1
ros2 run composition manual_composition

# Terminal 2:
ros2 node list
```
Gives:
```
$ ros2 node list
/internal_node
/listener
/talker
```

sloretz, 2022-11-03:
Does `/internal_node` still get renamed if it's node options get `use_global_arguments` set to false? I suspect it's picking up the name remap rule that was intended for the component manager.

https://github.com/ros2/rclcpp/blob/7c6785176a878cf16eb98995b732f405b9e776ed/rclcpp/include/rclcpp/node_options.hpp#L150-L160

methylDragon, 2022-11-04:
> Does `/internal_node` still get renamed if it's node options get `use_global_arguments` set to false? I suspect it's picking up the name remap rule that was intended for the component manager. [...]

It does not :o
Is there a way to make this the default behavior?

closed 2023-02-15 by methylDragon