ros2/rclcpp #2045: adress sanitizer complains new-delete-type-mismatch on allocator_common

tsingakbar, 2022-11-08:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 2204
- Installation type:
  - build from humble branch
- Version or commit hash:
  - [ae8b033ae01601475104bf9e4c2350b98c9560c4](https://github.com/ros2/rclcpp/commit/ae8b033ae01601475104bf9e4c2350b98c9560c4)
- DDS implementation:
  - any one(note related to rmw)

#### Steps to reproduce issue
build humble according to official guide enabling address sanitizer, like this:
```bash
colcon build --packages-up-to examples_rclcpp_minimal_publisher --cmake-args -DCMAKE_BUILD_TYPE=Debug -DBUILD_TESTING=OFF -DCMAKE_C_FLAGS=-fno-omit-frame-pointer -fsanitize=address -DCMAKE_CXX_FLAGS=-fno-omit-frame-pointer -fsanitize=address
```
Then launch the shipped example after source installed setup.bash:
`RMW_IMPLEMENTATION=rmw_fastrtps_cpp ASAN_OPTIONS=detect_leaks=0 install/examples_rclcpp_minimal_publisher/lib/examples_rclcpp_minimal_publisher/publisher_lambda`

#### Expected behavior
no address sanitizer problems.

#### Actual behavior
Then address sanitizer reports new-delete-type-mismatch, which is always related to https://github.com/ros2/rclcpp/blob/humble/rclcpp/include/rclcpp/allocator/allocator_common.hpp#L67 .

```c++
template<typename Alloc> void * retyped_allocate(size_t size, void * untyped_allocator);
template<typename T, typename Alloc> void retyped_deallocate(void * untyped_pointer, void * untyped_allocator);
```
Mostly `retyped_allocate` is called with a size greater than 1, but later calling `retyped_deallocate` will internally always use 1 as size to dealloc.

#### Additional information

fujitatomoya, 2022-11-08:
related to https://github.com/ros2/rclcpp/pull/1324, duplicate of https://github.com/ros2/rclcpp/issues/1925

tylerjw, 2022-11-12:
Here is my attempt at rebasing the stale PR created to fix this bug. It would be wonderful to be able to use ASAN with ROS.

https://github.com/ros2/rclcpp/pull/2046

roncapat, 2026-09-14:
Shall this one be closed?

jmachowinski, 2026-09-14:
Can't be fixed in humble but is fixed with lyrical onwards

closed 2026-09-14 by jmachowinski