ros2/rclcpp #2051: Wrong link to Humble API

jrutgeer, 2022-11-28:
The link to the rclcpp API for humble, as mentioned in the README file, is incorrect.

API used to be at [https://docs.ros2.org/latest/api/](https://docs.ros2.org/latest/api/) but now is at [https://docs.ros.org/en/humble/p/](https://docs.ros.org/en/humble/p/).

I was going to create a pull request to correct the link, but actually the better fix would probably be that [https://docs.ros2.org/latest/api/](https://docs.ros2.org/latest/api/) actually points to the latest api. But I don´t know where to file an issue for that.

fujitatomoya, 2022-12-01:
I think now the latest should be https://docs.ros.org/en/rolling/p/ instead of https://docs.ros.org/en/humble/p/? and i believe that this can be updated with appropriate one.
as you mentioned, it would be nice if https://docs.ros2.org/latest/api/ is proxied to https://docs.ros.org/en/rolling/p/ .

@clalancette what do you think?

jrutgeer, 2022-12-01:
@fujitatomoya I assumed 'latest' refers to 'latest released' and not 'latest development', but that might indeed be a false assumption.

In any case the current links are a mess:
- Google search on 'rclcpp api' yields links to https://docs.ros2.org/,
- The rclcpp readme also mentions https://docs.ros2.org/,
but:
- https://docs.ros2.org/<distro>/api/rclcpp/ exists only up to Galactic, not for Humble and Rolling,
- https://docs.ros2.org/latest/api/rclcpp/ seems to point to Foxy (!) API,

It's trivial in hindsight, but it took me a long time to find out that the link to a package API can be found on the package page at https://index.ros.org/.

hidmic, 2022-12-02:
I can attest we have some documentation discoverability issues. This is not unique to `rclcpp` though. Hoisting ticket to https://github.com/ros2/ros2.