ros2/rclcpp #2057: LifeCycle pattern dosn't work (for me at least)

DrShIkIgAmy, 2022-12-06:
Here's the problem:
I have LifeCycleNode implemented as component. The node has to received some data from a service to pass Configuration state. I bet you've already got the point. The configuration func is deadlocked. Because I can't return control to a component_container, cause configuration is not finished. I can't finish configuration, cause I've not received the data from the service (stiil awaiting). To receive response some of spins has to be called, but I can't do this, because my node will be double-registrated in that case. I cannot create another node (for client purpose) and destroy it after response is received, because any new node inside component derive name from parent. I cannot create sub-node, cause my node is derived from LifeCycleNode.
May be messed up a little bit, but it seems to me, that is is a bug.

fujitatomoya, 2022-12-06:
@DrShIkIgAmy thanks for creating issue, but i am not really sure about the case that you described. could you provide self-contained reproducible sample to this problem you have?

tgroechel, 2023-06-13:
If I'm parsing this correctly @DrShIkIgAmy , are you referring to the deadlock issue of calling a service within a running callback? I.e., https://github.com/ros2/rclcpp/issues/773 

Issue is:
- `on_change_state` calls a user callback (e.g., `on_configure`)
- within `on_configure` you call an external service
- the lc executor is stuck servicing this `on_configure` and can therefor not process any response (i.e., deadlocked)

This is likely addressed by https://github.com/ros2/rclcpp/issues/2213

fujitatomoya, 2023-06-15:
@tgroechel thanks for the info, https://github.com/ros2/rclcpp/issues/773#issuecomment-525939937 could be something we can do right now in the application.