ros2/rclcpp #2060: Missing nodelet_topic_tools headers

daisukes, 2022-12-09:
## Feature request

#### Feature description
- migrating [nodelet_topic_tools headers](https://github.com/ros/nodelet_core/tree/noetic-devel/nodelet_topic_tools/include/nodelet_topic_tools) or equivalent alternative
  ```
  nodelet_demux.h
  nodelet_lazy.h
  nodelet_mux.h
  nodelet_throttle.h
  ```
- related issues/discussions
  - https://github.com/ros-perception/image_pipeline/issues/780
  - https://github.com/ros-perception/perception_pcl/pull/385#discussion_r1041549677
    - perception_pcl uses demux, lazy, and mux
  - https://answers.ros.org/question/286069/implementing-subscriberstatuscallback-within-ros2/
  - https://www.robotandchisel.com/2020/07/29/five-things-ros2/#subscriber-connect-callbacks

#### Implementation considerations
- maybe in `include/rclcpp/topic_tools` or any better place/way?
- currently, polling is needed for lazy subscription - [workaround sample](https://github.com/mikeferguson/ros2_cookbook/blob/main/rclcpp/workarounds.md)
- this enhancement would enable us to use subscription callback like ROS1
  - https://github.com/ros2/rmw/issues/330

fujitatomoya, 2022-12-09:
@daisukes can you enlighten me more details about what the application want to do with example? i do not quite understand what is really required here...

> this enhancement would enable us to use subscription callback like ROS1

especially why this is related to this feature?

daisukes, 2022-12-09:
@fujitatomoya
Sure, one example is ROS2 porting of the `perception_pcl` package, which utilizes these headers.
Thanks to the introduction of type adaptation with Humble, I think many resource-intensive nodelets (image_pipeline, pointcloud_to_laserscan, etc) will be ported to ROS2 component, and many of them inherit classes in the `nodelet_topic_tools` headers, especially `nodelet_lazy.h`.

As for the lazy subscription (which is used in nodelte_lazy.h) it needs the feature (https://github.com/ros2/rmw/issues/330) to archive the functionality without polling the subscription match count.

`rclcpp_component` (equivalent to ROS1 nodelet) is in this repo, so, it would be great if these headers are in the common include.

Does this answer your questions?