ros2/rclcpp #2062: spin_some() exec one callback in one cycle

LANB0, 2022-12-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 
- Installation type:
  - binaries 
- Version or commit hash:
  - Foxy
- DDS implementation:
  - Fast-RTPS and cyclonedds
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
1.run the code
#include <time.h>
#include <chrono>  
#include <iostream> 
#include <rclcpp/rclcpp.hpp>
#include <can_msgs/msg/frame.hpp>
int main(int argc, char** argv)
{
    rclcpp::init(argc, argv);
    auto node = rclcpp::Node::make_shared("test");
    rclcpp::Rate loop_rate(20);

    auto subscriber = node->create_subscription<can_msgs::msg::Frame>("/test/can", 
        100, [](const std::shared_ptr<can_msgs::msg::Frame> msg){
            std::chrono::time_point<std::chrono::system_clock,std::chrono::milliseconds> tp = std::chrono::time_point_cast<std::chrono::milliseconds>(std::chrono::system_clock::now());//获取当前时间点
            std::time_t timestamp =  tp.time_since_epoch().count(); 
            std::cout<<"timestamp:"<<timestamp<<std::endl;
         });
    while (rclcpp::ok())
    {
        rclcpp::spin_some(node);
        loop_rate.sleep();
    }
    return 0;
}

2.use topic pub
ros2 topic echo -r 1000  /test/can can_msgs/msg/Frame "header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: ''
id: 100
is_rtr: false
is_extended: false
is_error: false
dlc: 8
data:
- 0
- 0
- 0
- 0
- 0
- 0
- 0
- 0"
```
#### Expected behavior
The code output the timestamp value 1ms interval.
#### Actual behavior
The code output the timestamp value 50ms interval. In 1s,the callback exec for 20 times.

#### Additional information
None

fujitatomoya, 2022-12-12:
> ros2 topic echo -r 1000  /test/can can_msgs/msg/Frame "header:

incorrect command line. it should be `ros2 topic pub ...`.

> The code output the timestamp value 50ms interval. In 1s,the callback exec for 20 times.

according to the sample above, this is correct and expected result.

- `std::chrono::duration` is set to `50 msec`, and it also be set into `rclcpp::Rate`.
-  `loop_rate.sleep();` will `50 msec` sleep every time single task (this case subscription callback) is executed by executor.

fujitatomoya, 2022-12-12:
@LANB0 i will go ahead to close this issue, if you have any comments, feel free to reopen or register new issue.

closed 2022-12-12 by fujitatomoya

LANB0, 2022-12-13:
> * `loop_rate.sleep();`每次执行器执行单个任务（本例订阅回调）时都会休眠。`50 msec`

Sorry for the error cmd.

> > ros2 主题回显 -r 1000 /test/can can_msgs/msg/Frame “header：
> 
> 不正确的命令行。应该是.`ros2 topic pub ...`
> 
> > 代码输出时间戳值 50ms 间隔。在1s内，回调执行20次。
> 
> 根据上面的示例，这是正确的和预期的结果。
> 
> * `std::chrono::duration`设置为 ，并且也设置为 。`50 msec``rclcpp::Rate`
> * `loop_rate.sleep();`每次执行器执行单个任务（本例订阅回调）时都会休眠。`50 msec`

> incorrect command line. it should be `ros2 topic pub ...`.
>
>
> according to the sample above, this is correct and expected result.
>
> * `std::chrono::duration` is set to `50 msec`, and it also be set [...]

Sorry for the error cmd.So the function spin_some() only execute the callback once,even if the msg queue has 1000 items?In ROS1,spinOnce() seems to get all the msgs and execute all callback in queue.