ros2/rclcpp #2068: Race condition in client async_send_request causing message loss

koonpeng, 2022-12-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu2204
- Installation type:
  - Both Binary and Source
- Version or commit hash:
  - humble
- DDS implementation:
  - default
- Client library (if applicable):
  - rclcpp

When using `async_send_request` with multi threaded executor, messages are sometimes lost and there appears to be no attempt to do retransmission (according to cyclonedds logs).

#### Steps to reproduce issue

Client code using multi threaded executor
```cpp
#include <example_interfaces/srv/add_two_ints.hpp>
#include <rclcpp/rclcpp.hpp>

#include <cinttypes>

using AddTwoInts = example_interfaces::srv::AddTwoInts;

int main(int argc, char* argv[])
{
  rclcpp::init(argc, argv);

  auto node = rclcpp::Node::make_shared("minimal_client");
  auto cb_group = node->create_callback_group(rclcpp::CallbackGroupType::MutuallyExclusive);
  using example_interfaces::srv::AddTwoInts;
  auto client = node->create_client<AddTwoInts>("add_two_ints", rmw_qos_profile_services_default, cb_group);

  auto req = std::make_shared<AddTwoInts::Request>();
  req->a = 1;
  req->b = 2;

  while (!client->wait_for_service(std::chrono::seconds(1))) {
    if (!rclcpp::ok()) {
      RCLCPP_ERROR(
        node->get_logger(),
        "client interrupted while waiting for service to appear.");
    }
    RCLCPP_INFO(node->get_logger(), "waiting for service to appear...");
  }

  auto timer = node->create_wall_timer(std::chrono::milliseconds(10), [&]() {
    auto fut = client->async_send_request(req);
    auto result = fut.get();
    RCLCPP_INFO(node->get_logger(), "result of %" PRId64 " + %" PRId64 " = %" PRId64, req->a, req->b, result->sum);
  });

  rclcpp::executors::MultiThreadedExecutor executor;
  executor.add_node(node);
  executor.spin();

  return 0;
}
```

* Run the server, pinning to cpu core 1 and 2. `taskset 0x2 ros2 run demo_nodes_cpp add_two_ints_server`.
* Run the client, `taskset 0x2 build/rclcpp_race_repro/rclcpp_race_repro`.

In my testing it reproduces reliably with 1-2 cores, as the core count increases it reproduces less frequently.

#### Expected behavior

All messages are received.

#### Actual behavior

Messages are lost and no retransmission is attempted.

#### Additional information

I managed to trace down the cause of the bug, it appears to be in

https://github.com/ros2/rclcpp/blob/f9050cd6663ccde9c4b6cabc8c7fa3ad154879ce/rclcpp/include/rclcpp/client.hpp#L793-L808

Because `pending_requests_mutex_` is not locked before sending the request, it is possible for the receiver thread to receive a response BEFORE the request is added to the pending list, in that case the receiver thread will silently drop it. The result is that the response is "lost", but on the RMW layer it is received and acknowledged so there is no re-transmission even if the qos is set to reliable.

clalancette, 2023-01-05:
> Is there any chance for it to be back ported?

So we can backport it to Humble, assuming that it doesn't break API or ABI.  If it does, then we'll need to find a different but equivalent solution.  If you could take a look and figure out the commit that fixed it, that would help in determining how difficult the backport is going to be.

fujitatomoya, 2023-01-06:
@clalancette there has been already one https://github.com/ros2/rclcpp/pull/2076, and we can keep ABI.

koonpeng, 2023-01-06:
Thanks, I think we can close this issue now.

closed 2023-01-06 by koonpeng