ros2/rclcpp #2073: [feature] easier user control over action goal id's

Aposhian, 2022-12-29:
## Feature request

#### Feature description
Currently `async_send_goal` simply calls `generate_goal_id`, which is just random bytes. It would be nice to have the following capabilities nicely supported:
- generate goal ID's that are well-formed UUIDs
- supply specific goal ID's (that come from other semantic contexts)

#### Implementation considerations
Here are a few ideas:
- provide an overloaded signature for `async_send_goal` that also includes a goal uuid parameter. Or as a default argument
- Add a goal UUID to the send goal options.
- Be able to set a "Generator" class/function on the action client class that generates UUIDs according to a certain scheme.

fujitatomoya, 2023-01-06:
> generate goal ID's that are well-formed UUIDs

this makes sense to me.If that is UUID, well-formed UUID would be useful especially for debugging or finding the UUID format via grep from the log.

> supply specific goal ID's (that come from other semantic contexts)

can you elaborate the use case for this extension? I am not really sure what could be beneficial for user application to generate and set the UUID for action goal...

Aposhian, 2023-01-06:
The benefit for setting a user-supplied goal ID is that if that goal ID carries semantics from somewhere else in the application.

Also, I found a simple workaround to support this sort of use case for now, simply by subclassing `rclcpp_action::Client<>` and then overriding `generate_goal_id`, since it is declared `virtual` in `rclcpp_action::ClientBase`. So that maybe is sufficient to support the "user supplied goal id" use case.

However, I think it would be good to just change the default implementation to generate UUIDv4 ids instead of random bytes.

CursedRock17, 2023-05-16:
When it comes to making the UUIDs more enforced, issue #1999 was solved to make the strings more compliant with RFC [4122](https://datatracker.ietf.org/doc/html/rfc4122). Was that the sort of thing we were looking to do make make these more set towards UUID V4, if so that part doesn't need to change. Otherwise, would we be aiming at making a method to be cryptographically ensured - the way boost does it - or more so based on a frequency timer that the RFC documentation goes through. To extent on this, @hidmic seems to have started some sort of method in the generate_goal_id method, was there any progress made in this?

As for allowing the user to provide input to the goal_id function, I created an override method which also takes a const GoalUUID& argument. This is a good idea because it will ensure that there are not duplicate UUIDs by some absurd offchance, if the user is using the same generator.