ros2/rclcpp #2083: it takes too long to terminate when using component_container_isolated

dujeong, 2023-01-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries (humble)
- Version or commit hash:
  - 
- DDS implementation:
  - cyclonedds
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I set `use_composition` to "True" to run the navigation stack nodes in one process and it worked.

after that, I typed `ctrl+c` to exit the process. but it takes too long to terminate(more than 10 seconds), and it does not terminate properly sometimes.

Below is part of the launch file I ran.

```
......

nav2_container = Node(
        condition=IfCondition(use_composition),
        package='rclcpp_components',
        executable='component_container_isolated',
        name=container_name,
        output='screen',
        respawn=use_respawn,
        respawn_delay=2.0,
        parameters=[configured_params_for_compositon, {'autostart': autostart}],
        remappings=remappings),
            
......

load_composable_nodes = LoadComposableNodes(
        condition=IfCondition(use_composition),
        target_container=container_name, #nav2_container
        composable_node_descriptions=[
            ComposableNode(
                package='nav2_controller',
                plugin='nav2_controller::ControllerServer',
                name='controller_server',
                parameters=[configured_params],
                remappings=[('cmd_vel', 'cmd_vel_nav')] + remappings),

            ComposableNode(
                package='nav2_smoother',
                plugin='nav2_smoother::SmootherServer',
                name='smoother_server',
                parameters=[configured_params],
                remappings=remappings),

            ComposableNode(
                package='nav2_planner',
                plugin='nav2_planner::PlannerServer',
                name='planner_server',
                parameters=[configured_params],
                remappings=remappings),

            ComposableNode(
                package='nav2_behaviors',
                plugin='behavior_server::BehaviorServer',
                name='behavior_server',
                parameters=[configured_params],
                remappings=remappings),

            ComposableNode(
                package='nav2_waypoint_follower',
                plugin='nav2_waypoint_follower::WaypointFollower',
                name='waypoint_follower',
                parameters=[configured_params],
                remappings=remappings),

            ComposableNode(
                package='nav2_bt_navigator',
                plugin='nav2_bt_navigator::BtNavigator',
                name='bt_navigator',
                parameters=[configured_params],
                remappings=remappings),

            ComposableNode(
                package='nav2_velocity_smoother',
                plugin='nav2_velocity_smoother::VelocitySmoother',
                name='velocity_smoother',
                parameters=[configured_params],
                remappings=[('cmd_vel', 'cmd_vel_nav'),
                            ('cmd_vel_smoothed', 'cmd_vel')] + remappings),

            ComposableNode(
                package='nav2_lifecycle_manager',
                plugin='nav2_lifecycle_manager::LifecycleManager',
                name='lifecycle_manager_navigation',
                parameters=[{'use_sim_time': use_sim_time,
                            'autostart': autostart,
                            'node_names': lifecycle_nodes}],
                remappings=remappings),
        ],
    )
    
.......
```

Below is the log when component container ends. (actually I use two container node, one is for navigation stack, and the other is for my custom node)

```
[ERROR] [component_container_isolated-13]: process[component_container_isolated-13] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[ERROR] [component_container_isolated-7]: process[component_container_isolated-7] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [component_container_isolated-13]: sending signal 'SIGTERM' to process[component_container_isolated-13]
[INFO] [component_container_isolated-7]: sending signal 'SIGTERM' to process[component_container_isolated-7]
[component_container_isolated-13] [INFO 1673835779.167809326] [rclcpp]: signal_handler(signum=15) (./src/rclcpp/signal_handler.cpp:71)
[component_container_isolated-7] [INFO 1673835779.169543075] [rclcpp]: signal_handler(signum=15) (./src/rclcpp/signal_handler.cpp:71)
[ERROR] [component_container_isolated-13]: process[component_container_isolated-13] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[ERROR] [component_container_isolated-7]: process[component_container_isolated-7] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [component_container_isolated-13]: sending signal 'SIGKILL' to process[component_container_isolated-13]
[INFO] [component_container_isolated-7]: sending signal 'SIGKILL' to process[component_container_isolated-7]
[ERROR] [component_container_isolated-7]: process has died [pid 22074, exit code -9, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_isolated --ros-args -r __node:=perception_container -r __ns:=/'].
[ERROR] [component_container_isolated-13]: process has died [pid 22086, exit code -9, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_isolated --ros-args -r __node:=nav_container -r __ns:=/ --params-file /tmp/tmpf4tgn9jt --params-file /tmp/launch_params_ei4e54qx -r /tf:=tf -r /tf_static:=tf_static'].

```

On the other hand, when using `component_container` instead of `component_container_isolated`, it is almost finished immediately.

Is it normal to take a long time to end when using `component_container_isolated`?

SteveMacenski, 2023-01-17:
Is this a problem with the container or Nav2? Do you have this issue if you use different components in the `component_container_isolated` container not in Nav2?

mjcarroll, 2023-01-17:
I actually believe this may be an issue with `component_container_isolated` as I have seen it in a non-navigation2 related test case

Since it is related to something I'm working on, I'll see if I can get a cleaner reproduction.

dujeong, 2023-01-18:
I also had the same problem when using non-Nav2 component. I am grateful that you are concerned about this matter.
I'll continue to watch the progress with interest. Thanks.

mjcarroll, 2023-01-18:
I now have a minimal reproduction example using a component with a single timer and publication (https://github.com/safe-ros/reference_system/blob/437dd76e4e61d9cb446bd9a55955d12a808be8ba/src/publisher_nodes.cpp#L22-L27)

The trace with `component_container_isolated` is:

```
[DEBUG] [1674052414.520708182] [rclcpp]: signal_handler(): notifying deferred signal handler
[DEBUG] [1674052414.520760484] [rclcpp]: deferred_signal_handler(): woken up due to SIGINT/SIGTERM or uninstall
[DEBUG] [1674052414.520779058] [rclcpp]: deferred_signal_handler(): shutting down
[DEBUG] [1674052414.520833875] [rclcpp]: deferred_signal_handler(): shutting down rclcpp::Context @ 0x5555555e1670, because it had shutdown_
on_signal == true
[DEBUG] [1674052414.520848901] [rcl]: Shutting down ROS client library, for context at address: 0x5555555733b0
[DEBUG] [1674052414.520864642] [rcl]: Calling timer
[DEBUG] [1674052414.520891271] [rcl]: Finalizing publisher
[Thread 0x7fffd3a25640 (LWP 2028681) exited]
[DEBUG] [1674052414.521215870] [rcl]: Publisher finalized
[DEBUG] [1674052414.521229548] [rcl]: Finalizing publisher
[DEBUG] [1674052414.521413831] [rcl]: Publisher finalized
[DEBUG] [1674052414.521448300] [rclcpp]: deferred_signal_handler(): waiting for SIGINT/SIGTERM or uninstall
```

fujitatomoya, 2023-01-18:
@iuhilnehc-ynos can you take a look at this, if you have time? (CC: @Barry-Xu-2018 )

Barry-Xu-2018, 2023-01-19:
@fujitatomoya We will enter Chinese New Year holiday from tomorrow. So we will continue to investigate it after holiday.

mjcarroll, 2023-01-19:
With a bit more debugging, I believe I have isolated it to a difference in the way that the [deferred_signal_handler](https://github.com/ros2/rclcpp/blob/37adc03c11171f67c213ee72f7ce43fdcbda7b9e/rclcpp/src/rclcpp/signal_handler.cpp#L258-L285) works between the isolated and non-isolated component containers.

In the isolated case, we end up blocking on a `sem_wait` and the semaphore never gets posted to because `Signalhandler::uninstall` never seems to get called.

A trace with more points added: https://gist.github.com/mjcarroll/a83c9b8428d198a2a7b753b19446f593

closed 2023-01-23 by mjcarroll

tonynajjar, 2023-08-08:
I spent 2 hours debugging this on humble before finally finding this fix. Thanks for the fix! Can you please backport to humble? I cherrypicked it locally and it worked well

PR #2085 (2023-01-23): Improve component_manager_isolated shutdown
This eliminates a potential hang when the isolated container is being shutdown via the rclcpp::SignalHandler.

Previously, when the SignalHandler triggers the shutdown of the rest of the system, the ComponentManagerIsolated object would cancel and destroy the executor associated with each loaded component.

Part of the cancel process checks if the executor is currently spinning as there may have been a race condition where `cancel_executor` is called before the thread is created.  While I didn't personally observe this condition, the comment indicate [... cut]