ros2/rclcpp #2096: rclcpp::Context::shutdown docs imply on_shutdown callbacks are called in the order added, but implementation iterates and unordered_set

mmattb, 2023-02-01:
## Bug report

**Required Info:**

- Operating System:
  - All
- Installation type:
  - All
- Version or commit hash:
  - be8c5d01c683ea381ffb77f4fc246411d8d31639   (i.e. today's HEAD)
- DDS implementation:
  - All
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

This is a rough sketch of what we are trying in nav2.

```
class MyChild : public rclcpp::LifecycleNode {
   MyChild() {
       register_my_rcl_preshutdown_callback();
   }

  ...
};

class MyDescendant : public MyChild
{
  MyDescendant() : MyChild(blah) {

    // Super constructor already ran by this point, so the MyDescendant
    // instance should have its preshutdown callback called before my_member's
    // preshutdown callback, but we are sometimes seeing them in reverse order!
    auto my_member = std::make_shared<MyChild>(blah);

  }

  ...
};
```

#### Actual behavior

In nav2 we are working on a fix involving the preshutdown callback mechanism. A pair of callbacks are sometimes firing in the reverse order from which they were added.

It looks like both preshutdown callbacks and on shutdown callbacks use an unordered_set:

https://github.com/ros2/rclcpp/blob/37adc03c11171f67c213ee72f7ce43fdcbda7b9e/rclcpp/include/rclcpp/context.hpp#L379

And they are executed via an iteration of those sets, which I believe means they will execute in some arbitrary hash order?

https://github.com/ros2/rclcpp/blob/37adc03c11171f67c213ee72f7ce43fdcbda7b9e/rclcpp/src/rclcpp/context.cpp#L315

#### Expected behavior

But the docs state pretty clearly that callbacks will be executed in the order they are added:

https://github.com/ros2/rclcpp/blob/37adc03c11171f67c213ee72f7ce43fdcbda7b9e/rclcpp/include/rclcpp/context.hpp#L228

https://github.com/ros2/rclcpp/blob/37adc03c11171f67c213ee72f7ce43fdcbda7b9e/rclcpp/include/rclcpp/context.hpp#L178

Seems like the docs are mismatched to the implementation. Unfortunately it may be annoying to reproduce since this involves a hash function... Might need to make a custom hash function to reproduce it.

#### Additional information

Can we please also add another line to the Context::shutdown docs regarding the preshutdown callbacks? Thanks to the people who added that callback; super handy!

SteveMacenski, 2023-02-01:
Maybe a dumb question, but why not just use something ordered `std::set` or `std::vector`?

clalancette, 2023-02-01:
> Maybe a dumb question, but why not just use something ordered `std::set` or `std::vector`?

It used to be a `std::vector`, but was changed to a `std::unordered_set` in #1639 (based on a comment of mine).  The basic reason is that when removing callbacks, we have to iterate through the entire list looking for the specific one to remove, which could be a performance problem.

That said, you could argue that removal isn't a common operation, so we should prioritize being consistent here.  I could buy that argument.  @Barry-Xu-2018 what do you think?

SteveMacenski, 2023-02-01:
Unfortunately we have to remove it due to some oddities with the shutdown procedures: https://github.com/ros-planning/navigation2/pull/3377#discussion_r1091031116

> This is to prevent a use after free in the case that the callback is called after the destructor fires. At that point the instance has evaporated, so there is no 'this' to bind the call to -> seg fault. I found this out the hard way :)

Basically, the preshutdown callbacks don't consider if the object registered even still exists (which for components, they can be unloaded or removed at any time) so the system seg faults. The preshutdown callback needs to be removed when we destroy our nodes to deal with that issue. I suppose the bigger issue is that we seg fault in the first place (ticket to follow) but either way, the shutdown callbacks should be executed deterministically when I add potentially many of them in a single node at different levels of object composition. 

Though, this is only really a problem we functionally run into because other libraries are having issue on clean shutdown. The docs say that it should be ordered, however. So if that's not the case and not intended to be fixed, the docs should be updated to reflect it

fujitatomoya, 2023-02-02:
I would also add test case for this.

Barry-Xu-2018, 2023-02-02:
If we only use std::set, I think it cannot make sure the order same as registered order.  

Reference to ParameterEventCallback, it uses std::unordered_map and std::list to implement.  std::list is used to make sure read order is the same as register order.  

https://github.com/ros2/rclcpp/blob/7d660acc05df6fc480d2b05e09d487d8c027d2c1/rclcpp/include/rclcpp/parameter_event_handler.hpp#L335-L342  

Considering the number of shutdown callback isn't large, std::vector is enough. The performance overhand for add and remove may be ignored (While adding, scan container to avoid repeat. While deleting, scan container to find deleted item.)

@clalancette agree with your thought.  

I will start to fix this bug.

fujitatomoya, 2023-02-06:
@mmattb this has been addressed by https://github.com/ros2/rclcpp/pull/2097, can we close this issue?

mmattb, 2023-02-07:
@fujitatomoya I haven't tested it, but it looks directionally correct. And it comes complete with unit tests! I'm fine with closing this.

SteveMacenski, 2023-02-07:
Thanks for the quick turn around!

closed 2023-02-07 by SteveMacenski