ros2/rclcpp #2103: Latest Doxygen docs does not seem to include all files

russkel, 2023-02-14:
#### Expected behavior
Documentation of classes such as `GenericSubscription` should be in the documentation at https://docs.ros2.org/latest/api/rclcpp/

#### Actual behavior
Classes don't appear at all - not even in the class index: https://docs.ros2.org/latest/api/rclcpp/classes.html#letter_g

jrutgeer, 2023-02-27:
docs.ros2.org is deprecated. It does no longer point to the latest API info.
Links to the latest API (for any package, not just rclcpp) can be found from the corresponding package info page at the ROS Index. For rclcpp it's at https://docs.ros.org/en/humble/p/rclcpp/

The rclcpp README was fixed to reflect this info: see PR #2071 

If you prefer the standard doxygen theme, I don´t know of another way than to download the source and manually run Doxygen, as described [here](https://discourse.ros.org/t/plans-for-ros-2-documentation-particularly-api-docs/28638/13). It's less of a hassle than you´d might expect.

fujitatomoya, 2023-02-27:
related to https://github.com/ros2/ros2_documentation/pull/3323

russkel, 2023-02-27:
> docs.ros2.org is deprecated. It does no longer point to the latest API info.

I didn't even realise there were two domains! Thanks @jrutgeer for enlightening me.

In which case, I change the point of this ticket, can we get docs.ros2.org to redirect to docs.ros.org?

clalancette, 2023-02-28:
> In which case, I change the point of this ticket, can we get docs.ros2.org to redirect to docs.ros.org?

Once Foxy goes out of support in May, we will do this.