ros2/rclcpp #2106: Documentation issue: Context documentation refer to non-existing rclcpp::init_local function

traversaro, 2023-02-16:
The documentation of Context contains (see https://github.com/ros2/rclcpp/blob/01b19247f10d1743d4ca96839283fdd40ac5fe4c/rclcpp/include/rclcpp/context.hpp#L68):
> It is often used in conjunction with rclcpp::init, or rclcpp::init_local,

However, unless I am missing something, I could not find any `rclcpp::init_local` function in the codebase.

## Bug report

**Required Info:**

- Operating System:
  - Not relevant.
- Installation type:
  - Not relevant.
- Version or commit hash:
  - 01b19247f10d1743d4ca96839283fdd40ac5fe4c
- DDS implementation:
  - Not relevant
- Client library (if applicable):
  - Not relevant

#### Steps to reproduce issue

Visit https://github.com/ros2/rclcpp/blob/rolling/rclcpp/include/rclcpp/context.hpp#L68

#### Expected behavior

Do not find reference to functions that do not exist.

#### Actual behavior

There is a reference to a function that does not exist.

clalancette, 2023-02-16:
Yes, I don't know how that got there; as far as I can tell, we've never had an `init_local` function.  Would you mind submitting a PR removing it?

traversaro, 2023-02-16:
> Yes, I don't know how that got there; as far as I can tell, we've never had an `init_local` function. Would you mind submitting a PR removing it?

Done in https://github.com/ros2/rclcpp/pull/2107 .

closed 2023-02-16 by clalancette