ros2/rclcpp #2110: Add callback for initalization of loaded components in rclcpp_components

mandicLuka, 2023-02-21:
## Feature request

#### Feature description

As the title suggests, the callback function for initialization of loaded component would be very beneficial. A lot of ros2 APIs use references or pointers to node, which are not accessible during the node construction. The recommended way now is to create init() method on the rclcpp::Node derivation and call this function manually from the node. 

#### Implementation considerations
As I can see, this would not be too much work to do, just the NodeFactoryTemplate class extension with one additional method that actually binds to template init() function. Then, ComponentManager can call this function right after node instantiation.

mandicLuka, 2023-02-21:
Following this question:
https://stackoverflow.com/questions/87372/check-if-a-class-has-a-member-function-of-a-given-signature

I have implemented the solution in the node_factory_template.hpp file.

The code first checks if node implements the init() method using template metaprogramming and then calls it if it exists.

Full code:

```
#ifndef RCLCPP_COMPONENTS__NODE_FACTORY_TEMPLATE_HPP__
#define RCLCPP_COMPONENTS__NODE_FACTORY_TEMPLATE_HPP__

#include <functional>
#include <memory>
#include <type_traits>

#include "rclcpp_components/node_factory.hpp"

namespace rclcpp_components
{

template<typename, typename T>
struct has_init {
    static_assert(
        std::integral_constant<T, false>::value,
        "Second template parameter needs to be of function type.");
};

// specialization that does the checking

template<typename C, typename Ret, typename... Args>
struct has_init<C, Ret(Args...)> {
private:
    template<typename T>
    static constexpr auto check(T*)
    -> typename
        std::is_same<
            decltype( std::declval<T>().init( std::declval<Args>()... ) ),
            Ret    // ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        >::type;  // attempt to call it and see if the return type is correct

    template<typename>
    static constexpr std::false_type check(...);

    typedef decltype(check<C>(0)) type;

public:
    static constexpr bool value = type::value;
};

/// NodeFactoryTemplate is a convenience class for instantiating components.
/**
 * The NodeFactoryTemplate class can be used to provide the NodeFactory interface for
 * components that implement a single-argument constructor and `get_node_base_interface`.
 */
template<typename NodeT>
class NodeFactoryTemplate : public NodeFactory
{
public:
  NodeFactoryTemplate() = default;
  virtual ~NodeFactoryTemplate() = default;

  /// Create an instance of a component
  /**
   * \param[in] options Additional options used in the construction of the component.
   */
  NodeInstanceWrapper
  create_node_instance(const rclcpp::NodeOptions & options) override
  {
    auto node = std::make_shared<NodeT>(options);

    if (has_init<NodeT, void(void)>::value)
    {
      node->init();
    }

    return NodeInstanceWrapper(
      node, std::bind(&NodeT::get_node_base_interface, node));
  }
};
}  // namespace rclcpp_components

#endif  // RCLCPP_COMPONENTS__NODE_FACTORY_TEMPLATE_HPP__

```

mandicLuka, 2023-02-22:
I have tested one possible solution in node_factory_template.hpp, following this question:
https://stackoverflow.com/questions/87372/check-if-a-class-has-a-member-function-of-a-given-signature

The solution uses template metaprogramming to check whether the templated node has an init() function and calls it after node creation.

Here is the full code:

```
#ifndef RCLCPP_COMPONENTS__NODE_FACTORY_TEMPLATE_HPP__
#define RCLCPP_COMPONENTS__NODE_FACTORY_TEMPLATE_HPP__

#include <functional>
#include <memory>
#include <type_traits>

#include "rclcpp_components/node_factory.hpp"

namespace rclcpp_components
{

template<typename, typename T>
struct has_init {
    static_assert(
        std::integral_constant<T, false>::value,
        "Second template parameter needs to be of function type.");
};

// specialization that does the checking

template<typename C, typename Ret, typename... Args>
struct has_init<C, Ret(Args...)> {
private:
    template<typename T>
    static constexpr auto check(T*)
    -> typename
        std::is_same<
            decltype( std::declval<T>().init( std::declval<Args>()... ) ),
            Ret    // ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        >::type;  // attempt to call it and see if the return type is correct

    template<typename>
    static constexpr std::false_type check(...);

    typedef decltype(check<C>(0)) type;

public:
    static constexpr bool value = type::value;
};

/// NodeFactoryTemplate is a convenience class for instantiating components.
/**
 * The NodeFactoryTemplate class can be used to provide the NodeFactory interface for
 * components that implement a single-argument constructor and `get_node_base_interface`.
 */
template<typename NodeT>
class NodeFactoryTemplate : public NodeFactory
{
public:
  NodeFactoryTemplate() = default;
  virtual ~NodeFactoryTemplate() = default;

  /// Create an instance of a component
  /**
   * \param[in] options Additional options used in the construction of the component.
   */
  NodeInstanceWrapper
  create_node_instance(const rclcpp::NodeOptions & options) override
  {
    auto node = std::make_shared<NodeT>(options);

    if constexpr (has_init<NodeT, void(void)>::value)
    {
      node->init();
    }

    return NodeInstanceWrapper(
      node, std::bind(&NodeT::get_node_base_interface, node));
  }
};
}  // namespace rclcpp_components

#endif  // RCLCPP_COMPONENTS__NODE_FACTORY_TEMPLATE_HPP__

```

mjcarroll, 2023-03-03:
I understand what you are running into here, but I would propose an alternative approach that was not previously available with ROS 1 nodelets.

The design of rclcpp_components makes it so that you no longer have to explicitly inherit from `rclcpp::Node` in order to register as a component.  You just have to iplement a constructor that takes `NodeOptions` and `get_node_base_interface` as the minimum interface.  

There is a full example on how to do this here:
https://github.com/ros2/demos/blob/humble/composition/src/node_like_listener_component.cpp

In your specific case, you could do something like:

```
// Some ROS 2 Interface that takes a shared pointer to node
class SomeRos2Interface
{
public:
  SomeRos2Interface(const std::shared_ptr<rclcpp::Node> &node);
};

class NodeLike
{
public:
  NodeLike(const rclcpp::NodeOoptions & options)
  {
    node_ = std::make_shared<rclcpp::Node>("node_name", options);
    interface_ = std::make_shared<SomeRos2Interface>(this->node_);
  }

  rclcpp::node_interfaces::NodeBaseInterface::SharedPtr
  NodeLike::get_node_base_interface()
  {
    return this->node_->get_node_base_interface();
  }

private:
  rclcpp::Node::SharedPtr node_;
  std::shared_ptr<SomeRos2Interface>  interface_;
};

RCLCPP_COMPONENTS_REGISTER_NODE(NodeLike);
```

One question I have is which other components explicitly requires the `Node::SharedPtr`?  At this point, I think we have tried to migrate all interfaces to accept raw pointers or the equivalent node interfaces.  I think something taking a SharedPtr may be a bug or oversight if it is something from the ROS 2 core packages.

mandicLuka, 2023-03-04:
Yeah, with ros2 core components, everything checks out, but image_transport, as far as I can see, does not have the option to pass raw pointers.

Your proposed solution seems to do the trick. Thanks :)

closed 2023-03-13 by mjcarroll