ros2/rclcpp #2118: Consider exposing services to update Parameter Descriptor ranges at runtime

sea-bass, 2023-03-06:
#### Feature description
It's great that parameter descriptors exist so they can be used for validating values -- however, we have encountered a use case where it would be useful to be able to SET parameter descriptors at runtime in addition to at declaration time.

Our use case is specifically to have UI sliders for parameter values where users can pre-configure their desired min and max values at the UI level. In other words, it's just the bounds we 

#### Implementation considerations
Think of the [`rcl_interfaces`](https://docs.ros2.org/latest/api/rcl_interfaces/index-msg.html) types available.

* For parameter values, we have `/get_parameters` and `/set_parameters` services available
* For parameter descriptors, we have `/describe_parameters` for reading, but nothing for updating the descriptions should we want to.

So, would it be beneficial to expose a similar service / service type to update parameter descriptors at runtime? I understand if there are concerns about the string description, type, and read-onlyness changing, but our use case at least is only concerned with the ranges. So even a service that allows modifying these for floating-point / integer-valued parameters would go a long way.

Thoughts?

roncapat, 2023-03-06:
It may be wise to set bounds as "writable" from inside the code. I mean, it should be an opt-in feature to make some parameter-description-related stuff writable/reconfigurable at runtime.

fujitatomoya, 2023-03-06:
sounds reasonable for me.

- Extend `NodeParameterInterface` to take `Descriptor` argument to update the parameter in the node.
  - if the parameter is declared with `Descriptor` read-only, this is NOT allowed to change parameter description.
- Expose new service to update `Descriptor` from `ParameterClient.

btw, question just came up, that if `dynamic_typing` is enabled (default), that makes more sense to update `floating_point_range` and `integer_range` as well?