ros2/rclcpp #2120: Logger::set_level does not lock the logging mutex

jrutgeer, 2023-03-07:
Shouldn't `set_level()` also lock the global logging mutex?

https://github.com/ros2/rclcpp/blob/dbe555a3c3470b051c5854881d464965a1d6c8a7/rclcpp/src/rclcpp/logger.cpp#L111-L115

Since it writes to the rcutils hashmap, which has no locks (see also https://github.com/ros2/rcutils/issues/397).

See also `get_child()` which does lock:

https://github.com/ros2/rclcpp/blob/dbe555a3c3470b051c5854881d464965a1d6c8a7/rclcpp/src/rclcpp/logger.cpp#L71-L82

clalancette, 2023-03-07:
In short: probably, yes.

The longer answer is that I think there are two different problems in this area of code:
1.  `Logger::set_level` doesn't seem to operate like the rest of the `Logger` class.  In particular, the rest of the `Logger` class is manipulating things via `rcl_logging_rosout`, while this call is directly using `rcutils_logging`.  Maybe that is just a case where we need to add another function to `rcl_logging_rosout`, but I'm not even sure of that.  Looking at this design again, I'm not quite sure why everything here is manipulating `rcl_logging_rosout`; the logging is not `rosout`-specific.  So there may be a deeper problem here.
2. The global logging mutex seems to be a bit confused on what it wants to protect.  It is currently used in these places:
    * `NodeBase::NodeBase` to lock around `rcl_node_init` (which ends up calling `rcl_logging_rosout_init_publisher_for_node`)
    * `rclcpp_logging_output_handler` to lock around `rcl_logging_multiple_output_handler`
    * `Context::init` to lock around `rcl_logging_configure_with_output_handler`
    * `Logger::get_child` to lock around `rcl_logging_rosout_add_sublogger`

These uses are all for logging stuff, true, but it seems like they span both regular logging and the rosout logging.  It's not totally clear to me what the design here should be, but I do think it needs a bit of work.  Just my 2 cents.

jrutgeer, 2023-03-07:
>Looking at this design again, I'm not quite sure why everything here is manipulating rcl_logging_rosout; the logging is not rosout-specific. So there may be a deeper problem here.

Well, from an rclcpp point of view, you cannot think of console logging, rosout logging and external library logging as separate things, as they are called consecutively on each log call, and multiple log calls can be made concurrently from different threads.
So in that aspect it makes sense that there is one global rclcpp logging mutex. Though it could be an optimization to have three.

Any writing call to logging is locked:

> `NodeBase::NodeBase` to lock around `rcl_node_init`
- Upon creating a node, the **rosout hashmap** needs to be updated, so it contains the link for that node name to the publisher,
  -  Hence the lock around rcl_node_init,

> `rclcpp_logging_output_handler` to lock around `rcl_logging_multiple_output_handler`
- Log calls from rclcpp are mutually exclusive
  -  Hence this lock

> `context::init` to lock around `rcl_logging_configure_with_output_handler`
- You can have multiple contexts, but logging is initialized only once (per [this check](https://github.com/ros2/rclcpp/blob/1fd5a96561166ad2ad557bbc1a297cd652883cb2/rclcpp/src/rclcpp/context.cpp#L218)).
  -  Concurrent calls to `init` should not be possible as there is an [init_lock](https://github.com/ros2/rclcpp/blob/1fd5a96561166ad2ad557bbc1a297cd652883cb2/rclcpp/src/rclcpp/context.cpp#L198),
  - And logging should not be done before `init` is finished, except in case of [bugs](https://github.com/ros2/rcl/issues/1037) maybe. ;-)
  - So debatable whether this lock is mandatory, but it does not harm I suppose, 
- Note that this call initializes all three logging systems: [here](https://github.com/ros2/rcl/blob/bc979c6f22d6af54ad473cd65cd81708a4da8a93/rcl/src/rcl/logging.c#L65) (which actually is a [bug](https://github.com/ros2/rcl/issues/1036)), [here](https://github.com/ros2/rcl/blob/4eccc3cbe65f5b58981f22efaf612a65731cb2f0/rcl/src/rcl/logging.c#L96) and [here](https://github.com/ros2/rcl/blob/bc979c6f22d6af54ad473cd65cd81708a4da8a93/rcl/src/rcl/logging.c#L103).

> `Logger::get_child` to lock around `rcl_logging_rosout_add_sublogger`
- On request of a child logger, the **rosout hashmap** needs to be updated so the child logger name is linked to the same publisher as the parent,
  -  Hence this lock

So currently any write access to any logging is mutually excluded by a mutex lock, **except** for the Logger::set_level. Hence this issue.

Though, now that I am writing this, I just realized the following: 

Read access of the loglevel, through `rcutils_logging_logger_is_enabled_for(name, severity)`, currently does *not* have any locks around it. What happens in case of a concurrent call with `rcutils_logging_set_logger_level`? Currently this is not likely, as loglevels are only set on initialization through the command line arguments, but [this](https://github.com/ros2/ros2/issues/1355#issuecomment-1456856756) might change that...

If this also needs a lock, then I think:
- Either the lock should be moved from the rclcpp multiple output handler to the rclcpp log macros (i.e. around the whole `if logger_is_enabled_for() { rcutils_log_internal();}` call, or
- Locking should be done at the level of rcutils, rosout and the external library.

First option is the easiest, and (without thinking it through) I think this would also fix [this](https://github.com/ros2/rcutils/pull/393).

Currently there only is a mutex in the external log library, not in rcutils and rosout.

jrutgeer, 2023-03-07:
> What happens in case of a concurrent call with rcutils_logging_set_logger_level?

The same issue as [rcutils #393](https://github.com/ros2/rcutils/pull/393) I suppose.

</br>

> Either the lock should be moved from the rclcpp multiple output handler to the rclcpp log macros

Though that would mean that a log macro call from a low priority thread could block a call from a higher priority thread, even when disabled through the loglevel.

clalancette, 2023-03-08:
> Currently there only is a mutex in the external log library, not in rcutils and rosout.

Correct.  Up until now, the design philosophy has always been to have the locking in the user-facing client libraries (rclcpp, rclpy, etc).  This was both a decision to keep the lower layers simpler, as well as to make it easier (as it both C++ and Python have cross-platform locking solutions).

However, because of the dependency-injection style of design here at the logging layer, it doesn't really work for logging.  We really do need to solve ros2/rcutils#397 .

clalancette, 2023-03-14:
> * Children of named loggers (e.g. `rclcpp::get_logger("parent.child")`) don't have the optimization, since you can't call `get_child()` on a named logger. But that is not a common use case (and could be discouraged in the documentation),

This situation is more common than you might think.  In particular, when a user creates an action server, for instance, we create children named loggers for some of the internal things.

Separately, if we are going to do this work for rclcpp, we'll also need to do the same for rclpy for the same reasons.

All of that said, I do think that this is an easier and more viable way forward than trying to change rcutils.  It improves the situation (and performance) for many common cases, and makes us thread-safe where we currently aren't.

So with that said, @jrutgeer are you up for trying an implementation?  I'm happy to do a review.

jrutgeer, 2023-03-16:
@clalancette 

> This situation is more common than you might think. In particular, when a user creates an action server, for instance, we create children named loggers for some of the internal things.

I am not sure. I had checked that, and unless I am missing something, these are all node loggers, not named loggers.

> are you up for trying an implementation?

Not sure yet, it is hard for me to assess how long it would take me. I haven't written any code since 2006. :-)

Also I'm not sure yet about the best solution:

A readers-writer lock isn't ideal, because this still blocks on the reader side if it concurs with a write.  Ideally it should be possible to use logging macros in realtime code during testing, and disable them through the loglevel for normal use. But that implies a non-blocking read of the loglevel.

So what about following alternative:

- Upon construction of an rclcpp::Logger, the loglevel is read from the hashtable (with read lock) and cached locally,
  - If there is no loglevel in the hashtable, the cached value is set to the default loglevel,
- Upon log macro call, `rcutils_logging_get_logger_effective_level(name)` is called only when a `try_lock()` succeeds,
   - If the `try_lock()` fails, the cached loglevel value is used instead,
   - Upon success, the cached loglevel is updated if needed.

A further optimization could be _not_ to call `rcutils_logging_get_logger_effective_level(name)` for node loggers, but always use the cached value. Since loglevel can only be changed upon initialization (i.e. before the node constructor) or through the  [upcoming service call](https://github.com/ros2/ros2/issues/1355), which could also update the cached value. The annoying thing is that you can call that service with the name of another node. In that case only the hashtable would be updated and not the other node's local variable. So the service call would work when called with:
- its own node name,
- any named logger's name,
- but not with another node's name.

Also in the case of multiple nodes with the same name it would only work for the node that executes the service call.

jrutgeer, 2023-03-17:
According to [this](https://www.youtube.com/watch?v=ndeN983j_GQ&t=879s), `std::mutex::try_lock()` can fail spuriously, what in above case isn't really an issue (it'll just mean the cached value is used for that macro call, but on the next one it should succeed), but also that `std::mutex::unlock()` can block, which _is_ an issue for realtime.

They propose the use of a spinlock intead of a mutex.