ros2/rclcpp #2124: Improve documentation around non-Node derived components

mjcarroll, 2023-03-13:
The design of rclcpp_components makes it so that you no longer have to explicitly inherit from rclcpp::Node in order to register as a component. You just have to implement a constructor that takes `NodeOptions` and `get_node_base_interface` as the minimum interface.

There is a demo available here: https://github.com/ros2/demos/blob/humble/composition/src/node_like_listener_component.cpp

And some discussion on why you would want to use this here: https://github.com/ros2/rclcpp/issues/2110#issuecomment-1454228192

It would be helpful to turn this into a document that is more easily searchable/indexable.

leeminju531, 2023-04-09:
I am interested in contributing to this and would like to improve the documentation. there is anyone who already progressing this?

mjcarroll, 2023-04-10:
Not that I'm aware of.  Thanks for helping out!

CursedRock17, 2023-09-26:
Is there still a need to update documentation or was the pull request linked, strong enough?

clalancette, 2023-09-26:
I think what we put in above is good enough for now, so I'll close this.  We can always add more documentation later.  Thanks!

closed 2023-09-26 by clalancette