ros2/rclcpp #2130: Deadlock in rclcpp_action::Client when calling async_get_result and receiving a result simultaneously

eschembor-irobot, 2023-03-17:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
  - 
- Installation type: From source
  - 
- Version or commit hash: 5a172751068cfea0d441cc056b8a2dc213f4b20b
  - 
- DDS implementation: fastdds
  - 
- Client library (if applicable): rclcpp
  - 

#### Steps to reproduce issue

fujitatomoya, 2023-03-18:
@eschembor-irobot thanks for posting issue.

true, this is inverse lock order, this will be deadlock. report says it is Galactic, but rolling has the same problem.

CC: @iuhilnehc-ynos

fujitatomoya, 2023-03-18:
quick code scan makes me think,

https://github.com/ros2/rclcpp/blob/1a796b5515cde3d8b6a64f8c53c7f49d8a742d32/rclcpp_action/src/client.cpp#L327

do we need to mutex lock during callback is executing? what about take out `std::map<int64_t, ResponseCallback>` with `result_requests_mutex` lock, and execute the callback w/o lock?

iuhilnehc-ynos, 2023-03-18:
or make https://github.com/ros2/rclcpp/blob/6815022909086487f2b8dc14c0540b00d4cf61a0/rclcpp_action/include/rclcpp_action/client.hpp#L499 to only lock https://github.com/ros2/rclcpp/blob/6815022909086487f2b8dc14c0540b00d4cf61a0/rclcpp_action/include/rclcpp_action/client.hpp#L500-L502

iuhilnehc-ynos, 2023-03-18:
No ABI-breaking fix seems good.

fujitatomoya, 2023-04-04:
closing since all related PRs are merged. (rolling, humble and foxy)

closed 2023-04-04 by fujitatomoya

vishalgir04, 2024-03-08:
@fujitatomoya Did this get backported for galactic as well?

clalancette, 2024-03-08:
> @fujitatomoya Did this get backported for galactic as well?

No, Galactic was end-of-life before this fix came in.