ros2/rclcpp #2144: Nodeoptions to disable rosout did't work

Lannister-Xiaolin, 2023-03-30:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- ROS2 Version:
galactic

#### Steps to reproduce issue

Init node as below shows:
```
Node("test_package_server",  rclcpp::NodeOptions().arguments( {"--ros-args", "__disable-rosout-logs"}))
```

#### Expected behavior
No rosout in rqt console
#### Actual behavior
Still have rosout in rqt console
But if I run command as below. it works.
```
ros2 run test_package test_package_server --ros-args --disable-rosout-logs
```
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

iuhilnehc-ynos, 2023-03-31:
If you want to disable `rosout` log for a specific node, please use
https://github.com/ros2/rclcpp/blob/7aff0ffc56b8875ceb7af6e5beb832721fffed19/rclcpp/include/rclcpp/node_options.hpp#L167-L176

```patch
- Node("test_package_server",  rclcpp::NodeOptions().arguments( {"--ros-args", "__disable-rosout-logs"}))
+ Node("test_package_server",  rclcpp::NodeOptions().enable_rosout(false))
```

fujitatomoya, 2023-03-31:
with quick code scan, as implementation.

- `ros2 run test_package test_package_server --ros-args --disable-rosout-logs` works since this is global argument.
- Node argument does not seem to support `--ros-args --disable-rosout-logs`, it can be parsed, but not effectively checked or set to `rcl_node_init`. instead, it expects that user can use `NodeOptions::enable_rosout(bool enable_rosout)` method.

fujitatomoya, 2023-03-31:
btw, with my environment.

> Node("test_package_server",  rclcpp::NodeOptions().arguments( {"--ros-args", "__disable-rosout-logs"}))

this does not work,

but `Node("test_package_server",  rclcpp::NodeOptions().arguments( {"--ros-args", "--disable-rosout-logs"}))`

closed 2023-05-10 by Lannister-Xiaolin