ros2/rclcpp #2146: Topic still registered after publisher has been destroyed

achim-k, 2023-03-30:
## Bug report

**Required Info:**

- Operating System: `Ubuntu 22.04.2 LTS`
- Installation type: `binaries`
- Version or commit hash: `19.0.0-1jammy.20230131.024259`
- DDS implementation: `rmw_fastrtps_cpp`
- Client library (if applicable): `rclcpp`

#### Steps to reproduce issue
1. Build and run the executable below
2. Confirm with `ros2 topic info /pub_topic` that the publisher is still registered even though it was shut down

```cpp
#include <rclcpp/rclcpp.hpp>
#include <std_msgs/msg/string.hpp>

int main(int argc, char *argv[]) {
  rclcpp::init(argc, argv);

  auto n = rclcpp::Node::make_shared("pub_node");
  auto pub = n->create_publisher<std_msgs::msg::String>("pub_topic", 10);

  std::thread turnOffPubThread([&n, &pub]() {
    std::this_thread::sleep_for(std::chrono::seconds(1));
    RCLCPP_INFO(n->get_logger(), "Shutting down publisher...");
    pub.reset();
  });

  rclcpp::spin(n);
  rclcpp::shutdown();
  turnOffPubThread.join();
  return 0;
}
```

#### Expected behavior
Once the publisher is destroyed, the published topic should not be visible anymore with `ros2 topic list`

#### Actual behavior
The published topic is still registered after the publisher has been destroyed

fujitatomoya, 2023-03-31:
Sorry not a concrete solution but just some internals and comments. I believe that this behavior depends on rmw implementation. as described, share pointer is released. that means, publisher object has been freed on `pub.reset();`.

One thing that i confirmed with `rmw_fastrtps` and `rmw_connextdds` is because the executor is still spinning on that node. (internally waiting forever on `rmw_wait` until next executable.) probably this does not give a chance to related objects such as topic and publisher to be released or cleared in the implementation.

with cyclonedds, even after `rclcpp::shutdown()` is called, we can still see the both topic and publisher endpoint.

see https://github.com/fujitatomoya/ros2_test_prover/commit/6ebbb03661730bdbdeb7697a7825f83c9638d5b8 and using `while true; do ros2 topic info /pub_topic; sleep 1; done` command line to monitor how the ros2 information changes.

mjcarroll, 2023-04-13:
> One thing that i confirmed with rmw_fastrtps and rmw_connextdds is because the executor is still spinning on that node.

This is interesting because `rclcpp::shutdown` should have fired the `shutdown` guard condition, which causes the `rmw_wait` to return and should allow for all of the resources to be cleaned up accordingly.

Can you also try with `ros2 topic list --no-daemon` to confirm that we aren't holding some state in the CLI tools daemon?

fujitatomoya, 2023-04-13:
> Can you also try with ros2 topic list --no-daemon to confirm that we aren't holding some state in the CLI tools daemon?

good suggestion, i will try that out.

fujitatomoya, 2023-04-13:
@achim-k do you have any other questions or comments?

achim-k, 2023-04-13:
> This is interesting because `rclcpp::shutdown` should have fired the `shutdown` guard condition, which causes the `rmw_wait` to return and should allow for all of the resources to be cleaned up [...]

I do not understand how `rclcpp::shutdown` is relevant here. Publishers should be unregistered while my node is still running, not when it is shutdown. In my specific case I check with [get_publishers_info_by_topic](https://docs.ros2.org/galactic/api/rclcpp/classrclcpp_1_1Node.html#a08220daf1d66596012e858d19efd1212) and I see that the publisher is still registered even though the pointer to it has been destroyed.

fujitatomoya, 2023-04-13:
@achim-k can you take a look at https://github.com/fujitatomoya/ros2_test_prover/commit/6ebbb03661730bdbdeb7697a7825f83c9638d5b8 sample code with

> One thing that i confirmed with rmw_fastrtps and rmw_connextdds is because the executor is still spinning on that node. (internally waiting forever on rmw_wait until next executable.) probably this [...]

does that work for your case?

if we keep `spin`ing and update the entity, i think that we need to wake up `rmw_wait` in some ways.

> I do not understand how rclcpp::shutdown is relevant here.

it is not related.

achim-k, 2023-04-13:
> does that work for your case?

It does, your assumptions seems to be correct. What I could do to trigger the cleanup, is to create a timer that gets destroyed once it gets out of scope. That seems to work:

```cpp
std::thread turnOffPubThread([&n, &pub]() {
    std::this_thread::sleep_for(std::chrono::seconds(1));
    RCLCPP_INFO(n->get_logger(), "Shutting down publisher...");
    pub.reset();

    // New: Create a timer that gets immediately out of scope (should never fire)
    n->create_wall_timer(std::chrono::seconds(1), []() {});
  });
```

fujitatomoya, 2023-11-02:
@achim-k i will go ahead to close this in favor of https://github.com/ros2/rclcpp/issues/2146#issuecomment-1507474907. please feel free to reopen the issue if you have any other issues.

closed 2023-11-02 by fujitatomoya