ros2/rclcpp #2150: undefined reference to `__atomic_exchange_1'

Jpgig, 2023-03-31:
I am using the ros2/rclcpp package for PolarFireSOC yocto build and getting the below error:

[image]

Steps followed:
1) $ mkdir yocto-dev && cd yocto-dev
$ repo init -u https://github.com/polarfire-soc/polarfire-soc-yocto-manifests.git -b main -m default.xml
repo sync
repo rebase
. ./meta-polarfire-soc-yocto-bsp/polarfire-soc_yocto_setup.sh
MACHINE=icicle-kit-es bitbake mpfs-dev-cli

2) Added ROS2 Humble meta layer
git clone https://github.com/vmayoral/meta-ros -b honister-humble

3) updated bblayers.conf file to add ROS2 humble meta layers
build/conf/bblayers.conf
<my-project-path>/meta-ros/meta-ros2 \
 <my-project-path>/meta-ros/meta-ros2-humble \
 <my-project-path>/meta-ros/meta-ros-common \

4) also added following variables in the bblayers.conf file
# define the ROS 2 Yocto target release
ROS_OE_RELEASE_SERIES = "honister"

# define ROS 2 distro
ROS_DISTRO = "humble"

5) created demo-image-ros2.bb as below:
require mpfs-dev-cli.bb

SUMMARY = "An image including a bare-minimum installation of ROS 2 and including some basic pub/sub examples. It includes two DDS middleware implementations, FastDDS and Cyclone DDS"
DESCRIPTION = "${SUMMARY}"

inherit ros_distro_${ROS_DISTRO}
inherit ${ROS_DISTRO_TYPE}_image

ROS_SYSROOT_BUILD_DEPENDENCIES = " \
    ament-lint-auto \
    ament-cmake-auto \
    ament-cmake-core \
    ament-cmake-cppcheck \
    ament-cmake-cpplint \
    ament-cmake-export-definitions \
    ament-cmake-export-dependencies \
    ament-cmake-export-include-directories \
    ament-cmake-export-interfaces \
    ament-cmake-export-libraries \
    ament-cmake-export-link-flags \
    ament-cmake-export-targets \
    ament-cmake-gmock \
    ament-cmake-gtest \
    ament-cmake-include-directories \
    ament-cmake-libraries \
    ament-cmake \
    ament-cmake-pytest \
    ament-cmake-python \
    ament-cmake-ros \
    ament-cmake-target-dependencies \
    ament-cmake-test \
    ament-cmake-version \
    ament-cmake-uncrustify \
    ament-cmake-flake8 \
    ament-cmake-pep257 \
    ament-copyright \
    ament-cpplint \
    ament-flake8 \
    ament-index-python \
    ament-lint-cmake \
    ament-mypy \
    ament-package \
    ament-pclint \
    ament-pep257 \
    ament-pycodestyle \
    ament-pyflakes \
    ament-uncrustify \
    ament-xmllint \
    cmake \
    eigen3-cmake-module \
    fastcdr \
    fastrtps-cmake-module \
    fastrtps \
    git \
    gmock-vendor \
    gtest-vendor \
    pkgconfig \
    python-cmake-module \
    python3-catkin-pkg \
    python3-empy \
    python3 \
    python3-nose \
    python3-pytest \
    rcutils \
    rmw-implementation-cmake \
    rosidl-cmake \
    rosidl-default-generators \
    rosidl-generator-c \
    rosidl-generator-cpp \
    rosidl-generator-dds-idl \
    rosidl-generator-py \
    rosidl-parser \
    rosidl-runtime-c \
    rosidl-runtime-cpp \
    rosidl-typesupport-c \
    rosidl-typesupport-cpp \
    rosidl-typesupport-fastrtps-cpp \
    rosidl-typesupport-interface \
    rosidl-typesupport-introspection-c \
    rosidl-typesupport-introspection-cpp \
    foonathan-memory-vendor \
    libyaml-vendor \
"

IMAGE_INSTALL:append = " \
    ros-base \
    examples-rclcpp-minimal-action-client \
    examples-rclcpp-minimal-action-server \
    examples-rclcpp-minimal-client \
    examples-rclcpp-minimal-composition \
    examples-rclcpp-minimal-publisher \
    examples-rclcpp-minimal-service \
    examples-rclcpp-minimal-subscriber \
    examples-rclcpp-minimal-timer \
    examples-rclcpp-multithreaded-executor \
    examples-rclpy-executors \
    examples-rclpy-minimal-action-client \
    examples-rclpy-minimal-action-server \
    examples-rclpy-minimal-client \
    examples-rclpy-minimal-publisher \
    examples-rclpy-minimal-service \
    examples-rclpy-minimal-subscriber \
    demo-nodes-cpp \
    demo-nodes-cpp-rosnative \
    demo-nodes-py \
    cyclonedds \
    rmw-cyclonedds-cpp \
    tmux \
    python3-argcomplete \
    glibc-utils \
    localedef \
    rt-tests \
    stress \
    ${ROS_SYSROOT_BUILD_DEPENDENCIES} \
    opencv-staticdev \
    libstdc++ \
    libstdc++-dev \
"

IMAGE_LINGUAS = "en-us"
GLIBC_GENERATE_LOCALES = "en_US.UTF-8"

6) Build the image:
MACHINE=icicle-kit-es bitbake demo-image-ros2

Appreciate any help to resolve this issue.

fujitatomoya, 2023-03-31:
@Jpgig thanks for creating issue, but i am not sure yocto build is officially supported. what about creating issue on https://github.com/vmayoral/meta-ros/tree/honister-humble or originally https://github.com/ros/meta-ros (which does not seem to support humble...) ?

closed 2023-04-13 by Jpgig

robwoolley, 2025-01-21:
I have fixed this in meta-ros by explicitly adding -latomic to LDFLAGS for RISC-V board support packages: https://github.com/ros/meta-ros/commit/e4af3cf867c3fe30ac233f08f1aead777a975a47

I observed that most distros are resolving this missing libatomic problem for RISC-V (and others like MIPS and PPC, I believe) in the build system of individual packages as not all packages require libatomic.  The foonathan_memory package is a good example.  Just wanted to let you know in case it comes up again.