ros2/rclcpp #2161: Publish dynamic TF with this->get_node_topics_interface() works in foxy but not in humble.

BlueBirdHouse, 2023-04-07:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 22.04.2 LTS
- Installation type:
  binaries
- Version or commit hash:
  ROS humble
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue

Make a dynamic publisher as we always do in foxy.
```
rclcpp::node_interfaces::NodeTopicsInterface::SharedPtr node_topics_handle = this->get_node_topics_interface();

std::unique_ptr<tf2_ros::TransformBroadcaster> tf_broadcaster_ = nullptr;
tf_broadcaster_ = std::make_unique<tf2_ros::TransformBroadcaster>(node_topics_handle);
```

#### Expected behavior

The dynamic TF broadcaster object can be initialized. 

#### Actual behavior

It will give you an error.
"passed non-default qos overriding options without providing a parameters interface"

#### Additional information

[image]

schornakj, 2023-04-14:
This seems like it was probably a change in the `tf2_ros` API. You could probably get a more specific answer if you post an issue here: https://github.com/ros2/geometry2/issues

BlueBirdHouse, 2023-04-14:
Thanks, I have posted my question at the URL you suggested.

BlueBirdHouse, 2023-04-15:
I got an answer from [Url](https://github.com/ros2/ros2/issues/1407). Thanks anyway.

closed 2023-04-15 by BlueBirdHouse