ros2/rclcpp #2166: subsequent nodes in a namespace in yaml don't get parameters applied

Aposhian, 2023-04-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - humble
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Try a params yaml like this:
```
namespace:
  node1:
    ros__parameters:
      param: 1.0
  node2:
    ros__parameters:
      param: 2.0
```

#### Expected behavior

`node1` to have `param` with a value of 1.0, and `node2` to have `param` with a value of 2.0

#### Actual behavior

`node1` has a `param` set to 1.0 and `node2` has a value of `param` that is the default. The value from yaml is not applied.

#### Additional information
Also reported here: https://answers.ros.org/question/394231/ros2-how-to-get-multiple-nodes-under-same-namespace-in-yaml-config-file/

Interestingly, having top level duplicate keys works, even though that is not technically valid yaml, like this:

```
namespace:
  node1:
    ros__parameters:
      param: 1.0
namespace:
  node2:
    ros__parameters:
      param: 2.0
```

fujitatomoya, 2023-04-20:
using https://github.com/fujitatomoya/ros2_test_prover/blob/master/prover_rclcpp/src/rclcpp_2166.cpp with

```yaml
testns:
  node1:
    ros__parameters:
      param: 1.0
  node2:
    ros__parameters:
      param: 2.0
```

i do not have the problem with rolling https://github.com/ros2/ros2/commit/fea3f22ebc1becc6303aebcc9f93bde5abf0f92d

```bash
### Start 2 nodes under namespace testns with loading parameter file
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run prover_rclcpp rclcpp_2166 --ros-args --params-file test_params.yaml
[INFO] [1681970561.659209435] [testns.node1]: node1 created...
[INFO] [1681970561.661174465] [testns.node2]: node2 created...

# Check parameter list for all nodes
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param list
/testns/node1:
  param
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  use_sim_time
/testns/node2:
  param
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  use_sim_time
```

unfortunately i am not sure which PR addressed this issue. CC: @iuhilnehc-ynos

iuhilnehc-ynos, 2023-04-20:
> i do not have the problem with rolling [ros2/ros2@fea3f22](https://github.com/ros2/ros2/commit/fea3f22ebc1becc6303aebcc9f93bde5abf0f92d)
> unfortunately i am not sure which PR addressed this issue. CC: @iuhilnehc-ynos

I am not able to reproduce the issue either in `humble` or `rolling`. More information is needed.

fujitatomoya, 2023-04-20:
@Aposhian we are not able to reproduce the issue, can you confirm the reproducible environment?

Aposhian, 2023-04-20:
Ok I'll try to get a repro soon