ros2/rclcpp #2172: Apply a fix for LifecycleNode::get_current_state thread safety which is equivalent to #1756 into Humble

schornakj, 2023-04-25:
There is a thread safety bug in `LifecycleNode` which was originally documented in #1746 and fixed in #1756. However, the way the fix was implemented involves a breaking ABI change, and since it was merged after the Humble release it cannot be directly backported to Humble.

This is actually a quite severe bug, especially for systems which are using ros2_control's ControllerManager, which is tightly integrated with the lifecycle node concept. My context here is that I work with several robots which need to stay up and continuously running 24/7, and every few days I see a segfault like the ones posted in https://github.com/ros-controls/ros2_control/issues/979. From my perspective the fix for this certainly seems like something that should be backported into the LTS release as well.

From reading the discussion in #1756 and related PRs, one reason why a fix took so long to get merged and involved an ABI break is that the work to implement the fix uncovered several other bugs in LifecycleNode, and the fixes for those bugs needed to get merged as a precondition for merging #1756. If I want to apply an equivalent fix to Humble, I'll need to follow a different path since I need to preserve ABI compatibility. Does anyone who's already familiar with these libraries have ideas for how to approach this?

clalancette, 2023-04-27:
I'll leave a few comments:

* Backporting https://github.com/ros2/rclcpp/pull/2027 (one of the underlying PRs) is not going to be possible, as it breaks API and ABI.  So skip this one.
* Backporting https://github.com/ros2/rclcpp/pull/2031 (one of the underlying PRs) breaks ABI, so we can't backport it.  This one is minor anyway.
* That leaves backporting https://github.com/ros2/rclcpp/pull/1756 itself.  If you go back to my original comments from https://github.com/ros2/rclcpp/pull/1756#discussion_r708303467, my major concern here was returning an internal variable member from `get_current_state`.  We aren't going to be able to fix that one because of ABI problems, but it may be enough to just take the rest of that PR as it was at the time and leave the problematic `get_current_state` as it is.
* Note that even with the previous bullet point, we still have to maintain ABI in humble.  So, for instance, we can't add the new `state_handle_mutex_` member to the `State` class or the `state_machine_mutex_` to the `LifecycleNodeInterface` class.  One trick I've used in the past to get around this is to introduce a global map with a pointer to the class as the key and the mutex you want to use as the value.  Then when you want to lock in a function, you look up the mutex using `this` and then lock it.  Note that the global map also needs its *own* mutex to avoid multiple threads accessing it at once.

I hope this helps point in the correct direction.

schornakj, 2023-05-01:
> Note that even with the previous bullet point, we still have to maintain ABI in humble. So, for instance, we can't add the new state_handle_mutex_ member to the State class or the [...]

The bulk of the changes in #1756 are to `lifecycle_node_interface_impl.hpp` and `lifecycle_node_interface_impl.cpp`, and I'm wondering if the `LifecycleNodeInterfaceImpl` class which is defined in those files is really a part of the public interface? It's used as a private unique_ptr member with a forward declaration [here](https://github.com/ros2/rclcpp/blob/00ef09cbf34563db27b9aac9b0206eb142cf3995/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node.hpp#L988-L989) in `LifecycleNode`, and both the header and source file for `LifecycleNodeInterfaceImpl` are in `rclcpp_lifecycle/src`. That seems like a textbook example of the [pimpl pattern](https://en.cppreference.com/w/cpp/language/pimpl) commonly used to allow making internal changes to a class while preserving ABI compatibility. @clalancette am I thinking along the right track here?

clalancette, 2023-05-01:
> The bulk of the changes in #1756 are to `lifecycle_node_interface_impl.hpp` and `lifecycle_node_interface_impl.cpp`, and I'm wondering if the `LifecycleNodeInterfaceImpl` class which is defined in [...]

You are correct w.r.t. `state_machine_mutex_` in the `LifecycleNodeInterfaceImpl` class; thanks for pointing that out.  That part doesn't have an ABI problem, since it is all implementation.

However, we still have the ABI problem with `state_handle_mutex_` in the `State` class.  If we can figure out a way to preserve ABI there (ala my previous comments), then the rest of #1756 should be backportable.

schornakj, 2023-05-02:
@clalancette I followed your suggested strategy for protecting `state_handle_` in the `State` class and pushed up #2183. Let me know what you think!

schornakj, 2023-06-27:
#2183 got merged, so I'll close this as completed.

closed 2023-06-27 by schornakj