ros2/rclcpp #2180: `ExecutorEntitiesCollection` does not handle stale entities correctly

alsora, 2023-05-02:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu
- Installation type:
  -source
- Version or commit hash:
  - iron
- DDS implementation:
  - N/A
- Client library (if applicable):
  - rclcpp

I found a bug where the new `ExecutorEntitiesCollection` struct does not handle correctly stale entities.
The struct contains several maps of the form (simplified here for compactness)
```
struct TimerEntry
{
  rclcpp::TimerBase::WeakPtr entity;
  rclcpp::CallbackGroup::WeakPtr callback_group;
};

  std::unordered_map<const rcl_timer_t *, TimerEntry>
```

I have a test where I create, destroy and recreate timers multiple times (this is not the purpose of the test, it just happens).
Consider the following flow:
 - create timer and add it to the collection
 - timer goes out of scope -> now the TimerBase object stored in the collection is invalid
 - create timer again -> this new timer ends up having the same `rcl_timer_t *` as before, but it's a new rclcpp timer 

In this situation, when I go to refresh the collection, the new timer is not added because we only look at the rcl pointer key and it looks like the timer is already there.

I think that this is a corner case due to the OS reusing the same addresses for some objects, indeed if I add/remove some completely unrelated memory allocations in the test, the issue doesn't occurr.

Ideas for solutions:
 - Have the `update_entities` function explicitly handle expired weak pointers in the current collection
 - Require users of the entities collection class to perform a cleanup step once in a while

mjcarroll, 2023-05-02:
> I think that this is a corner case due to the OS reusing the same addresses for some objects, indeed if I add/remove some completely unrelated memory allocations in the test, the issue doesn't occurr.

This makes sense, I noticed a similar failure in a test that I'm sorting out in the waitset implementation, but hadn't had a chance to completely isolate it, so this is very helpful.

I like the idea of having update_entities culling the current collection, as it's less work on the user.  I will try an implementation in that direction.

alsora, 2023-05-02:
Ok, locally I'm using a dedicated function such as 
```
  void remove_stale_entities()
  {
    for (auto it = this->begin(); it != this->end(); ) {
      auto entity = it->second.entity.lock();
      if (entity) {
        ++it;
      } else {
        it = this->erase(it);
      }
    }
  }
```

but this can better be integrated into the update call (as I'm calling this new function on all collections before calling update)

clalancette, 2023-11-29:
Just as an FYI, https://github.com/ros2/rclcpp/pull/2142/commits/4fe9f68e8b6eacb219684e95bbab381780084b40 (part of #2142) should fix this.  Maybe not the most efficient way (as we are adding in another iteration over the collections), but it seems to fix the problems in my testing.