ros2/rclcpp #2216: Registration of multiple callbacks on state transition

LastStarDust, 2023-06-16:
## Feature request

#### Feature description

It would be nice if it was possible to register multiple callbacks to be called on a certain state transition.
The callbacks should be then called in the order of insertions.

For example, in the current implementation every time the method `register_on_activate(callback)` is called, the previous callback is overwritten.

#### Implementation considerations

This is because the callback container is implemented as a map.
See [lifecycle_node_interface_impl.hpp#L149-L151](https://github.com/ros2/rclcpp/blob/rolling/rclcpp_lifecycle/src/lifecycle_node_interface_impl.hpp#L149-L151):
```
  std::map<
    std::uint8_t,
    std::function<node_interfaces::LifecycleNodeInterface::CallbackReturn(const State &)>> cb_map_;
```

alsora, 2023-06-16:
This use case can easily be implemented on the application side (e.g. your mother node has a list of callbacks and it register a callback that iterates through them).

This also seems a better approach to me, since it allows you to hide the lifecycle details from the child nodes.

LastStarDust, 2023-06-16:
Moreover, the semantic of `register_on_configure` is vaguely evocative of a "callback addition" more than a "callback overwrite".

tgroechel, 2023-06-16:
@LastStarDust  I think this is a good idea in theory but agree with @alsora that this should probably live as application-side behavior.

> Nevertheless, I still think this feature would be useful. What if I do not want to hide the lifecycle details from the child nodes? What if I want the child node to be able to register callbacks for all the possible transitions?

I would argue:
1. default behavior would be to hide the lc details (sort of the point of lc in the first place in my opinion)
2. if the child were to register callbacks with the parent via user implementation (as opposed to the backend map), then the parent could (and should in my opinion) own the child callbacks / their ordering which will likely be context/application dependent. The parent would more likely be responsible for ordering of callback calls as opposed to insertion order (which is flimsy/difficult to re-order/could have constructor order issues depending on implementation)
3. having more than one callback in the lc backend (the map you were referring to) would be problematic. By default, the lc creates callbacks that just return `CallbackReturn::SUCCESS` so users don't need to implement boiler plate for transition functions they do not need. This would require a mechanism for ids for callbacks, how to prune them, when to do so etc

> Moreover, the semantic of the name register_on_configure is vaguely evocative of a "callback addition" more than a "callback overwrite".

This is a good point, possibly future consideration of renaming to `override_on_configure` or equivalent

LastStarDust, 2023-06-16:
@tgroechel Point taken. I am going to implement this at the application level. As you say, it takes a little effort but it has its merits to manage this at the application level.

LastStarDust, 2023-06-16:
Feel free to close this issue.

closed 2023-06-16 by alsora

fujitatomoya, 2023-06-16:
@alsora @LastStarDust @tgroechel 

> This also seems a better approach to me, since it allows you to hide the lifecycle details from the child nodes.
> ParentNode::register_post_on_configure_cb(std::function) with the ChildNode registering their callback here. 

i am not sure how this is related to multiple callbacks support? this sounds to me one of the application designs.
i mean even we have multiple callbacks supported in `LifecycleNode`, that can be application design if we want to? (i may be mistaken...)

i think that as `LifecycleNode` class itself, having multiple callbacks make senses and flexible for user application.

if a `LifecycleNode` node configures multiple sensors to configure `on_configure` transition, those functions would be provided sensor libraries.
those can be easily registered using multiple callback support without application wrapping those functions into single callback in the application code?

> By default, the lc creates callbacks that just return CallbackReturn::SUCCESS so users don't need to implement boiler plate for transition functions they do not need.

this still can be supported with multiple callbacks? if no callbacks are registered in the map, we can return `CallbackReturn::SUCCESS`?

> This would require a mechanism for ids for callbacks, how to prune them, when to do so etc

true.

besides, my major concern was `what if 1st callback succeeds and 2nd callback fails?` with multiple callbacks for a transition state.
probably we cannot roll back to previous primary state completely.
i believe that this is the case why it supports the single callback for each transition state.
i do not know the design history, so might be missing something.

what do you think?

tgroechel, 2023-06-16:
@fujitatomoya Thanks for the follow up, my thoughts below. *Apologies for large comment, I wish Github had threads.*

> > ParentNode::register_post_on_configure_cb(std::function) with the ChildNode registering their callback here.

> i am not sure how this is related to multiple callbacks support? this sounds to me one of the application designs.
i mean even we have multiple callbacks supported in LifecycleNode, that can be application design if we want to? (i may be mistaken...)

Ah wrote my response a bit quickly. I was meaning this to be an example of how a user *could* design their application.  I still think single callback is "better" design (see below).

> > This would require a mechanism for ids for callbacks, how to prune them, when to do so etc

> true.

> besides, my major concern was what if 1st callback succeeds and 2nd callback fails? with multiple callbacks for a transition state.
> probably we cannot roll back to previous primary state completely. [...]

I don't know the history but I would hedge this is the core reason (and better than my given reasons https://github.com/ros2/rclcpp/issues/2216#issuecomment-1594685442). 

I think a core idea is that the user is the only one who knows how to "rollback" a transition (unloading, flipping flags etc.) with it being the user's responsibility to return a `CallbackReturn::FAILURE`. The responsibility of the `LifecycleNode` backend, in my opinion, is to:
1. offer guarantees / manage underlying state+transitions logic
2. offer services (e.g., `GetState.srv`, `ChangeState.srv`) to introspect / request changes to the state
3. manage `ManagedEntities`
4. offer user transition flexibility via user defined transition functions (e.g., `on_configure`)

Therefor it is the user's responsibility to rollback anything that they may have done during a transition.

> i think that as LifecycleNode class itself, having multiple callbacks make senses and flexible for user application.

> if a LifecycleNode node configures multiple sensors to configure on_configure transition, those functions would be provided sensor libraries.
those can be easily registered using multiple callback support without application wrapping those functions into single callback in the application code?

I think this is likely a common design pattern (loading multiple dependencies, often external such as parameters & hardware), especially in `on_configure`.  I think it brings up 2 things that have come up recently:

1. loading these external dependencies often require calling a service within a service which is unsupported / blocking (outlined in https://github.com/ros2/rclcpp/issues/2057#issuecomment-1589186065 from today, @fujitatomoya thanks for the response). 

 I think a good way to address this is via adding response deferral functionality to lifecycle transitions: https://github.com/ros2/rclcpp/issues/2213

2. [Separate but related] It can be cumbersome to do a bulk set of service requests. Something possibly worth thinking about is creating a `BatchServiceRequester` utility that could send out a set of requests using a btach of client, moving `SharedFuture` callbacks to be waited on. It would be able to track all requests, their fail/success, and have a callback given a fail/success. I'm working on outlining something like this at the moment so this is still very much a theoretical idea / separate issue.