ros2/rclcpp #2217: heap-use-after-free on time_source.cpp

Rrooach, 2023-06-19:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- Installation type:
Source Code
- Version or commit hash:
Humble
- DDS implementation:
FastRTPS
- Client library (if applicable):
RCLCPP

#### Steps to reproduce issue
To reproduce this issue we need to install  ros2-humble with the Clang compiler, and with ASAN enabled, then run demo program  test_executor.
``` bash
colcon build --cmake-args  -DCMAKE_CXX_FLAGS="-g  -fsanitize=address"  -DCMAKE_C_FLAGS="-g -fsanitize=address "
```
After compiling, use objdump to check if the instrumentation is successful.
```bash
 objdump -t ./test_executor| grep asan
0000000000000000 l    df *ABS*	0000000000000000              asan_preinit.cpp
000000000027b9f6 g     O .bss	0000000000000001              __odr_asan._ZN7testing28FLAGS_gtest_break_on_failureE
0000000000000000       F *UND*	0000000000000000              __asan_report_load4
000000000027ad9d g     O .bss	0000000000000001              __odr_asan._ZN49TestExecutor_remove_callback_group_null_node_Test10test_info_E
0000000000000000       F *UND*	0000000000000000              __asan_stack_malloc_4
0000000000000000       F *UND*	0000000000000000              __asan_report_store4
0000000000000000       F *UND*	0000000000000000              __asan_report_store_n
000000000027ada1 g     O .bss	0000000000000001              __odr_asan._ZN54TestExecutor_constructor_bad_guard_condition_init_Test10test_info_E
000000000027ad97 g     O .bss	0000000000000001              __odr_asan._ZN43TestExecutor_spin_all_invalid_duration_Test10test_info_E
```
then this problem can be triggered.

#### Expected behavior
Program execute without error
#### Actual behavior
ASAN panic with the following console output:
https://pastebin.com/wdzRyJrZ

#### Additional information
I'm still new to ROS, but it seems that unchecked memory access on NodeBaseInterface, causes this UAF, it would be so nice of you that can help me figure this problem out.

fujitatomoya, 2023-06-20:
@Rrooach 

> Program execute without error

what program did you execute and observe [this asan stack trace](https://pastebin.com/wdzRyJrZ)?

Rrooach, 2023-06-21:
> what program did you execute

sorry for the ambiguous here, the program is a demo program called test_executor within ros2-humble source code, the execution path is /path/to/ros/build/rclcpp/test/rclcpp/test_executor

Before the execution, we may need to ```export  ASAN_OPTIONS=new_delete_type_mismatch=0``` to disable certain irrelevant information.

fujitatomoya, 2023-06-21:
thanks, this only happens on humble but rolling.

```bash
root@tomoyafujita:~/ros2_ws/humble_ws# source install/local_setup.bash
root@tomoyafujita:~/ros2_ws/humble_ws# export ASAN_OPTIONS=new_delete_type_mismatch=0
root@tomoyafujita:~/ros2_ws/humble_ws# build/rclcpp/test/rclcpp/test_executor
Running main() from /root/ros2_ws/humble_ws/install/gtest_vendor/src/gtest_vendor/src/gtest_main.cc

...<snip>
=================================================================
==729402==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 526080 byte(s) in 913 object(s) allocated from:
    #0 0x7fc1256d3c18 in __interceptor_realloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:164
    #1 0x7fc1240af90a  (/lib/x86_64-linux-gnu/libstdc++.so.6+0xb090a)

SUMMARY: AddressSanitizer: 526080 byte(s) leaked in 913 allocation(s).
```