ros2/rclcpp #2220: AddressSanitizer: new-delete-type-mismatch

fujitatomoya, 2023-06-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - source build `rolling`
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/fea3f22ebc1becc6303aebcc9f93bde5abf0f92d
- DDS implementation:
  - Any
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```bash
colcon build --symlink-install --cmake-args -DCMAKE_CXX_FLAGS="-g  -fsanitize=address" -DCMAKE_C_FLAGS="-g -fsanitize=address"
source install/local_setup.bash
ros2 run demo_nodes_cpp talker
...
```

#### Expected behavior

No ASAN report generated.

#### Actual behavior

The following ASAN report generated.

<details><summary>AddressSanitizer: new-delete-type-mismatch</summary>

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp talker
=================================================================
==726931==ERROR: AddressSanitizer: new-delete-type-mismatch on 0x60600006c0e0 in thread T0:
  object passed to delete has wrong type:
  size of the allocated type:   64 bytes;
  size of the deallocated type: 1 bytes.
    #0 0x7f7b840bd22f in operator delete(void*, unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:172
    #1 0x7f7b82772587 in rcutils_string_map_fini /root/ros2_ws/colcon_ws/src/ros2/rcutils/src/string_map.c:105
    #2 0x7f7b8354dfb0 in rcl_resolve_name /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/node_resolve_name.c:110
    #3 0x7f7b8354e751 in rcl_node_resolve_name /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/node_resolve_name.c:152
    #4 0x7f7b835483e5 in rcl_publisher_init /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:83
    #5 0x7f7b83c0bc16 in rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_message_type_support_t const&, rcl_publisher_options_s const&, rclcpp::PublisherEventCallbacks const&, bool) /root/ros2_ws/colcon_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/publisher_base.cpp:73
    #6 0x7f7b83a55937 in rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >::Publisher(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&) /root/ros2_ws/colcon_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/publisher.hpp:140
    #7 0x7f7b83a56e01 in void __gnu_cxx::new_allocator<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > >::construct<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&>(rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >*, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&) /usr/include/c++/11/ext/new_allocator.h:162
    #8 0x7f7b83a56e01 in void std::allocator_traits<std::allocator<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > > >::construct<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&>(std::allocator<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > >&, rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >*, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&) /usr/include/c++/11/bits/alloc_traits.h:516
    #9 0x7f7b83a56e01 in std::_Sp_counted_ptr_inplace<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >, std::allocator<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > >, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&>(std::allocator<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&) /usr/include/c++/11/bits/shared_ptr_base.h:519
    #10 0x7f7b83a56e01 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >, std::allocator<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&>(rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >*&, std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > > >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&) /usr/include/c++/11/bits/shared_ptr_base.h:650
    #11 0x7f7b83a56e01 in std::__shared_ptr<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > > >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&) /usr/include/c++/11/bits/shared_ptr_base.h:1342
    #12 0x7f7b83a56e01 in std::shared_ptr<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > >::shared_ptr<std::allocator<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > > >, rclcpp::node_interfaces::NodeBaseInterface*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&) /usr/include/c++/11/bits/shared_ptr.h:409
    #13 0x7f7b83a56e01 in std::shared_ptr<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > > std::allocate_shared<rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >, std [... cut ...]
    #1 0x7f7b82772023 in rcutils_string_map_init /root/ros2_ws/colcon_ws/src/ros2/rcutils/src/string_map.c:67
    #2 0x7f7b8354dc37 in rcl_resolve_name /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/node_resolve_name.c:47
    #3 0x610000000a3f  (<unknown module>)

SUMMARY: AddressSanitizer: new-delete-type-mismatch ../../../../src/libsanitizer/asan/asan_new_delete.cpp:172 in operator delete(void*, unsigned long)
==726931==HINT: if you don't care about these errors you may set ASAN_OPTIONS=new_delete_type_mismatch=0
==726931==ABORTING
[ros2run]: Process exited with failure 1
```

</details>

fujitatomoya, 2023-06-20:
this can also happen with humble branch as well.

fujitatomoya, 2023-06-20:
CC: @iuhilnehc-ynos @Barry-Xu-2018

iuhilnehc-ynos, 2023-06-21:
Actually, I noticed this issue a long time ago while adding the [zero_allocate](https://github.com/ros2/rclcpp/pull/1995).

- allocate
https://github.com/ros2/rclcpp/blob/978439191fdb3924af900a506dd35b68f8da725c/rclcpp/include/rclcpp/allocator/allocator_common.hpp#L40

- deallocate
https://github.com/ros2/rclcpp/blob/978439191fdb3924af900a506dd35b68f8da725c/rclcpp/include/rclcpp/allocator/allocator_common.hpp#L67

The `deallocate` just deallocate buffer size with 1. I don't know how to fix this issue elegantly.
Any suggestion?

iuhilnehc-ynos, 2023-07-10:
What about using an extra size in front of the buffer to store the size value when allocating a buffer?

CursedRock17, 2023-07-20:
According to [cppreference](https://en.cppreference.com/w/cpp/memory/allocator_traits/allocate) the ending parameter of both `allocate` and `deallocate` is a `size_type n` where n is:
> the number of objects to allocate storage for

So is there anyway to extract the size of the typename T given in `retyped_deallocate` like 
```
std::allocator_traits<Alloc>::deallocate(*typed_allocator, typed_ptr, sizeof(T));
```
or a some way to find the type of the object that was allocated, or is that beyond the reach of the function.

iuhilnehc-ynos, 2023-07-20:
What if the type is `char` and then `rcl` uses the allocator to allocate a buffer with `count`?
Even if we can add an explicit specialization `template get_rcl_allocator` for `char` to use `rcl_get_default_allocator()`, which seems like a workaround, what if using `get_rcl_allocator<rcl_node_t>` in the future, and then use the allocator to allocate a buffer with `count(2)`?

It seems that using `sizeof(T)` can't fix this issue.

ralwing, 2024-02-26:
There are many errors related to this issue:
* https://github.com/ros2/rclcpp/issues/1948
* https://github.com/ros2/rclcpp/issues/2217
* https://github.com/ros2/rclcpp/issues/2045
* https://github.com/autowarefoundation/autoware.universe/issues/3400

volleypaul, 2024-10-10:
Is this still an issue in Jazzy?  Thanks.

fujitatomoya, 2024-10-10:
yes, this issue is still open.

russkel, 2024-11-13:
So what is required for this to be fixed? I am running into some issues and would like to be able to run our stack with these sanitisers.

skndi, 2024-11-21:
Does anybody have a workaround for this problem? We're having some memory issues in one of our executables and I have a strong suspicion that the cause might be this bug. We're not using any custom allocators in our project so I would be okay even with replacing the body of the retyped_* functions with malloc/free/realloc.

wkaisertexas, 2024-12-13:
@kfabian This is still a problem on Linux though. I am getting the error as soon as a C++ node registers.  

```
[trajectory_planner_node-1] =================================================================
[trajectory_planner_node-1] ==1615252==ERROR: AddressSanitizer: new-delete-type-mismatch on 0x60200001fab0 in thread T0:
[trajectory_planner_node-1]   object passed to delete has wrong type:
[trajectory_planner_node-1]   size of the allocated type:   11 bytes;
[trajectory_planner_node-1]   size of the deallocated type: 1 bytes.
[trajectory_planner_node-1]     #0 0x7ffff74dc632 in operator delete(void*, unsigned long) (/usr/lib/llvm-14/lib/clang/14.0.0/lib/linux/libclang_rt.asan-x86_64.so+0xdc632) (BuildId: a6105a816e63299474c1078329a59ed80f244fbf)
[trajectory_planner_node-1]     #1 0x7ffff7f60f7e  (/opt/ros/humble/lib/librcl.so+0x21f7e) (BuildId: 7fd31c98930ca9fdbef0dc1d0891f1f556370d43)
[trajectory_planner_node-1]     #2 0x7ffff7f61229  (/opt/ros/humble/lib/librcl.so+0x22229) (BuildId: 7fd31c98930ca9fdbef0dc1d0891f1f556370d43)
[trajectory_planner_node-1]     #3 0x7ffff7f61d8e  (/opt/ros/humble/lib/librcl.so+0x22d8e) (BuildId: 7fd31c98930ca9fdbef0dc1d0891f1f556370d43)
[trajectory_planner_node-1]     #4 0x7ffff7f62107 in rcl_node_resolve_name (/opt/ros/humble/lib/librcl.so+0x23107) (BuildId: 7fd31c98930ca9fdbef0dc1d0891f1f556370d43)
[trajectory_planner_node-1]     #5 0x7ffff7f622d8 in rcl_publisher_init (/opt/ros/humble/lib/librcl.so+0x232d8) (BuildId: 7fd31c98930ca9fdbef0dc1d0891f1f556370d43)
[trajectory_planner_node-1]     #6 0x7ffff736228d in rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>> const&, rosidl_message_type_support_t const&, rcl_publisher_options_s const&) (/opt/ros/humble/lib/librclcpp.so+0x15828d) (BuildId: 1c49c0943977bdcb14f922706d316375108def41)
[trajectory_planner_node-1]     #7 0x7ffff732679b  (/opt/ros/humble/lib/librclcpp.so+0x11c79b) (BuildId: 1c49c0943977bdcb14f922706d316375108def41)
[trajectory_planner_node-1]     #8 0x7ffff7326ed0 in std::_Function_handler<std::shared_ptr<rclcpp::PublisherBase> (rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>> const&, rclcpp::QoS const&), rclcpp::PublisherFactory rclcpp::create_publisher_factory<rcl_interfaces::msg::ParameterEvent_<std::allocator<void>>, std::allocator<void>, rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void>>, std::allocator<void>>>(rclcpp::PublisherOptionsWithAllocator<std::allocator<void>> const&)::'lambda'(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>> const&, rclcpp::QoS const&)>::_M_invoke(std::_Any_data const&, rclcpp::node_interfaces::NodeBaseInterface*&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>> const&, rclcpp::QoS const&) (/opt/ros/humble/lib/librclcpp.so+0x11ced0) (BuildId: 1c49c0943977bdcb14f922706d316375108def41)
[trajectory_planner_node-1]     #9 0x7ffff731b7d3 in rclcpp::node_interfaces::NodeParameters::NodeParameters(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter>> const&, bool, bool, rclcpp::QoS const&, rclcpp::PublisherOptionsBase const&, bool, bool) (/opt/ros/humble/lib/librclcpp.so+0x1117d3) (BuildId: 1c49c0943977bdcb14f922706d316375108def41)
[trajectory_planner_node-1]     #10 0x7ffff730a71d in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>> const&, rclcpp::NodeOptions const&) (/opt/ros/humble/lib/librclcpp.so+0x10071d) (BuildId: 1c49c0943977bdcb14f922706d316375108def41)
[trajectory_planner_node-1]     #11 0x7ffff730af57 in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>> const&, rclcpp::NodeOptions const&) (/opt/ros/humble/lib/librclcpp.so+0x100f57) (BuildId: 1c49c0943977bdcb14f922706d316375108def41)
```

Am I reading [rclcpp/rclcpp/include/rclcpp/allocator/allocator_common.hpp](https://github.com/ros2/rclcpp/blob/978439191fdb3924af900a506dd35b68f8da725c/rclcpp/include/rclcpp/allocator/allocator_common.hpp#L67) correct in that it assumes that allocations are always of size 1? In glibc malloc throws away this information, but when using the address sanitizer or jemalloc or tcmalloc  this results in issues?

kfabian, 2024-12-13:
> @kfabian This is still a problem on Linux though. I am getting the error as soon as a C++ node registers.

You are not referring to the workaround to always use the result of `rcl_get_default_allocator()` or?

> Am I reading [...]

Basically yes. Not glibc malloc throws away the size, but the [libcstc++ implementation](https://github.com/gcc-mirror/gcc/blob/master/libstdc%2B%2B-v3/libsupc%2B%2B/del_ops.cc#L31) of `void operator delete  ( void*, std::size_t)`. jemalloc and I guess also tcmalloc and asan [overload the default implementation](https://github.com/jemalloc/jemalloc/blob/e4817c8d89a2a413e835c4adeab5c5c4412f9235/src/jemalloc_cpp.cpp#L205) with one actually using the size argument.

A proper solution might be to change the signature of the `rcutils_allocator_t` [deallocate](https://github.com/ros2/rcutils/blob/rolling/include/rcutils/allocator.h#L54) function to require a size argument, but that would be a huge effort to pass the size everywhere `deallocate` it is called.

wkaisertexas, 2024-12-13:
`lidstc++` only has `free(ptr)` in the method header right. So, any call made by the c standard library can only use that function which does not have a size argument. glibc's allocator does not throw away the size, it just does not get the size to begin with.

Using `std::allocator_traits<Alloc>::deallocate` library is used which requires a deallocation size even though the deallocation size is not known for thinks like C-style arrays.

I think this should at least have a ton of warnings in the [allocator template tutorial that the size of the deallocation may not be accurate](https://docs.ros.org/en/foxy/Tutorials/Advanced/Allocator-Template-Tutorial.html). Would you support a PR which added those changes?

kfabian, 2024-12-17:
> `lidstc++` only has `free(ptr)` in the method header right. So, any call made by the c standard library can only use that function which does not have a size argument. glibc's allocator does not throw away the size, it just does not get the size to begin with.

That is the reason why this problem only occurs if a malloc implementation like jemalloc or tcmalloc override `void operator delete  ( void*, std::size_t)` with a version that does not simply call `free` on the provided ptr, but makes actually use of the provided size for optimizations.

>Using std::allocator_traits<Alloc>::deallocate library is used which requires a deallocation size even though the deallocation size is not known for thinks like C-style arrays.

I think in practice it would be possible to provide the size to `rcutils_allocator_t::deallocate` because the container-like structs used in `rcl`, e.g. `rcutils_uint8_array_t` have a `buffer_capacity` member which could potentially be passed to the `deallocate` function.

jmachowinski, 2026-04-01:
https://github.com/ros2/rclcpp/pull/3058 fixed the reported issue. @mini-1235 the issue you are seeing are normal leaks which may be caused by the test you used. 

I am going to close this issue as resolved.