ros2/rclcpp #2225: Node crashes when calling services of same type loaded by plugins

pepeRossRobotics, 2023-06-23:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 22.04
- Installation type:
  binaries
- Version or commit hash:
  Humble
- DDS implementation:
  Cyclone
- Client library (if applicable):
  rclcpp

#### Expected behavior
When making a node that loads a bunch of plugins at run time. Each service should be independent from each other (when using different names).
Loading order of the plugins  should not have any effect on the execution of the services

#### Actual behavior
Depending on the Plugin load order, when calling certain services the Node (that loads the plugins) crashes.
Order that allows the services to work:
```
mk4_robot_interface:
  ros__parameters:
    plugins: [
      'EStopInterface',
      'PartyModeInterface',
      ]
```
Order that makes the node crash when calling a service of the Emergency Stop
```
mk4_robot_interface:
  ros__parameters:
    plugins: [
      'PartyModeInterface',
      'EStopInterface',
      ]
```
`[ERROR] [mk4_robot_interface-1]: process has died [pid 956612, exit code -11`
#### Additional information
Code to load the plugins:
```
for (uint8_t i = 0; i < plugins_to_load.size(); i++) {
        std::string plugin_name = "interface_plugin::" + plugins_to_load[i];
        try {
          auto plugin_instance = plugin_loader.createUnmanagedInstance(plugin_name);
          plugin_instance->initialize(plugin_info);
        } catch (pluginlib::PluginlibException & ex) {
          RCLCPP_WARN(
            get_logger(),
            "%s failed to load. Error: %s",
            plugin_name.c_str(),
            ex.what());
        }
```
The code that creates the respective services is:
```
set_party_mode_service = plugin_info.node->create_service<std_srvs::srv::SetBool>(
    plugin_info.node_name + "/set_party_mode",
    std::bind(
      &PartyModeInterface::set_party_mode_srv_callback,
      this,
      std::placeholders::_1,
      std::placeholders::_2,
      std::placeholders::_3),
      rmw_qos_profile_services_default,
      plugin_info.callback_group);
```
and for the other service
```
  set_e_stop_state_service_ = plugin_info.node->create_service<std_srvs::srv::SetBool>(
    plugin_info.node_name + "/set_e_stop_state",
    std::bind(
      &EStopInterface::set_e_stop_state_callback,
      this,
      std::placeholders::_1,
      std::placeholders::_2,
      std::placeholders::_3),
      rmw_qos_profile_services_default,
      plugin_info.callback_group);
```

Things to notice is that other plugins that have the same type of service have the same issues. Meaning that to load all the plugins of the project, I need to find the correct loading order to avoid crashing the node when calling services.

fujitatomoya, 2023-06-23:
@pepeRossRobotics is that possible if you can provide sample package can reproduce this issue? i think that would be helpful to get the help for this issue.

pepeRossRobotics, 2023-06-26:
I will prepare a package today and share it. Whats the best way to do this?

pepeRossRobotics, 2023-06-26:
[test_ws.zip](https://github.com/ros2/rclcpp/files/11865679/test_ws.zip)
This is an isolated bit of code that reproduces the problem. The do reproduce it, compile the ws then in a terminal run:
`ros2 launch mk4_robot_interface mk4_robot_interface.launch.py`
and on another terminal call the service:
`ros2 service call /mk4_robot_interface/set_party_mode std_srvs/srv/SetBool "data: true"`
You will notice this one succeeds. Then call 
`ros2 service call /mk4_robot_interface/set_e_stop_state std_srvs/srv/SetBool "data: true"` and you will notice the node crashes.
Modify the order in the config file for the plug in load order and re do it, the node does not crash. The problem is that later adding more plugins, the loading order becomes quite tricky.

iuhilnehc-ynos, 2023-06-26:
I think we need to make sure the class loader instance is live. e.g.,

```patch
diff --git a/include/mk4_robot_interface/mk4_robot_interface.hpp b/include/mk4_robot_interface/mk4_robot_interface.hpp
index e03a61c..d089322 100644
--- a/include/mk4_robot_interface/mk4_robot_interface.hpp
+++ b/include/mk4_robot_interface/mk4_robot_interface.hpp
@@ -37,6 +37,7 @@ public:
   }
 
 private:
+  std::unique_ptr<pluginlib::ClassLoader<robot_interface_driver::PluginInterface>> plugin_loader;
   rclcpp::TimerBase::SharedPtr initialize_plugins_timer;
   rclcpp::TimerBase::SharedPtr update_timer;
 
@@ -99,8 +100,8 @@ private:
   void initialize_plugins()
   {
     if (!are_plugins_initialized) {
-      pluginlib::ClassLoader<robot_interface_driver::PluginInterface> plugin_loader(
-        node_name, "robot_interface_driver::PluginInterface");
+      plugin_loader = std::make_unique<pluginlib::ClassLoader<robot_interface_driver::PluginInterface>>
+        (node_name, "robot_interface_driver::PluginInterface");
 
       RCLCPP_INFO(get_logger(), "Loading...");
       PluginInfo plugin_info;
@@ -113,7 +114,7 @@ private:
       for (uint8_t i = 0; i < plugins_to_load.size(); i++) {
         std::string plugin_name = "interface_plugin::" + plugins_to_load[i];
         try {
-          auto plugin_instance = plugin_loader.createUnmanagedInstance(plugin_name);
+          auto plugin_instance = plugin_loader->createUnmanagedInstance(plugin_name);
           plugin_instance->initialize(plugin_info);
         } catch (pluginlib::PluginlibException & ex) {
           RCLCPP_WARN(

```

pepeRossRobotics, 2023-06-26:
Thank you very much,  I think this fixes the issue. I will close this and will continue testing. If anything comes up, I will update

closed 2023-06-26 by pepeRossRobotics