ros2/rclcpp #2231: Add feedback to rejected action goal so the client can know why it was rejected

jlack1987, 2023-07-03:
## Feature request

#### Feature description

When an action goal is sent to a server the server can reject it, but there is no way for the server to tell the client why it was rejected. I think at the very least allowing a string description of why the goal was rejected so the other side can decide what to do next.

fujitatomoya, 2023-07-10:
I think ROS 2 only returns `true` (accepted) or `false` (refected) as underlying service `SendGoalService` internally. (saying user callback response is wrapped into the ROS service message.)

IMO, adding arbitrary message field would be useful for the application if we have full control for both service server and client implementation.

clalancette, 2023-07-20:
The original feature request seems reasonable; we'd just have to add a field to the messages that you would like to add this to.

There is a larger question of returning reject reasons to any service, but I think that is probably outside the scope of this issue.

Isaac-G5900, 2025-11-08:
Hey all, I was browsing through some issues and found that  I'd like to tackle this enhancement. 

After looking at the server.cpp file, I see that `ServerBase::execute_goal_request_received` checks acceptance cases via `GoalResponse::ACCEPT_AND_EXECUTE == status || GoalResponse::ACCEPT_AND_DEFER == status` and makes no use of the REJECT enum value to check a rejection case. I also noticed there was no 'call_goal_rejected_callback()' function in the server base implementation but I suppose there is no need since users already can know a goal was rejected(just not why). Adding that would require a much more involved change as well.

I thought that the path of least friction would be to add a check for GoalResponse::REJECT == status, enumerate some rejection cases as to why, and publish the appropriate reason message via the feedback publisher.  

This approach allows the following:
-Leverage existing infrastructure
-Allows users to optionally subscribe to reject reasons through the Feedback topic 
-Far less changes needed to accomplish tending to this issue

Let me know about any further insights or push back on this approach. 

Cheers