ros2/rclcpp #2247: Prefer custom exceptions to regular runtime_errors

CursedRock17, 2023-07-28:
## Feature request - Prefer custom exceptions compared to bulky if/else statements with `std::runtime_error`

#### Feature description
There are a few TODOs in the repository that are marked similar to:
> TODO(jacobperron): use custom exception

They can found by grepping for that format, like [here](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/src/rclcpp/node_interfaces/node_waitables.cpp#L35) where they are often surrounded by bulky if/else statements that have already been fixed to look better like [this](https://github.com/ros2/rclcpp/blob/rolling/rclcpp_components/src/component_manager.cpp#L115-#L122). There are most likely more of these style statements in the code base, that just haven't been marked, for instance the NodeWaitables example was recognized when the original pull request was made in 2018, but these were the easiest examples to target and start with.

#### Implementation considerations
I guess the main question is what the original authors meant by "custom" because in the [component_manager example](https://github.com/ros2/rclcpp/blob/rolling/rclcpp_components/src/component_manager.cpp#L115-#L122) there was a `ComponentManagerException` class:
```
class ComponentManagerException : public std::runtime_error
{
public:
  explicit ComponentManagerException(const std::string & error_desc)
  : std::runtime_error(error_desc) {}
};
```
It seems that it's a needless wrapper for `std::runtime_error` unless there's a benefit in passing the description via const reference. Would each of these exception follow a similar  format, could a new general class be made, or can we just stick with runtime_error?

clalancette, 2023-07-31:
> It seems that it's a needless wrapper for `std::runtime_error` unless there's a benefit in passing the description via const reference.

My understanding is that it is not a needless wrapper.  In particular, what it allows you to do is write code that catches rclcpp ComponentManager exceptions, but not all `std::runtime_error` exceptions.  Something like:

```
try {
    component_manager->do_something();
} catch (const rclcpp::ComponentManagerException & cme) {
    rclcpp_component_manager_error_handling();
} catch (const std::runtime_error & e) {
    general_error_handling();
}
```

CursedRock17, 2023-08-01:
Ok, I have also seen other classes with their own, little Exception handler class. In that case, can I open a PR that deals with all the current cases:
> TODO(person) use custom exception

To get their own Exception classes. Then, if needed deal with other areas that could maybe also use this style?

clalancette, 2023-08-01:
> In that case, can I open a PR that deals with all the current cases:

I'm going to suggest that you don't open a single PR that deals with all of them.  That is going to be difficult to review and get merged, since we have to inspect both the `throw` site as well as the handling stack.  Instead, I'll suggest you open a PR on a case-by-case basis, and we can examine each of them to make sure that is the thing we want to do.

closed 2023-08-04 by CursedRock17