ros2/rclcpp #2249: rclcpp needs specific version of rcl [humble]

stan-guer, 2023-07-28:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 22.04 
- Installation type:
  - binaries
- Version or commit hash:
  - ROS 2 Humble Hawksbill - Patch Release 4
- DDS implementation:
  - N/A
- Client library (if applicable):
  - rclcpp, rcl

#### Steps to reproduce issue

```
rocker --home --network host -- ros:humble-ros-base
apt update; apt install ros-humble-rclcpp # do not update ros-humble-rcl
# build any simple colcon package
```

#### Expected behavior

No errors

#### Actual behavior

Getting error:
```
/usr/bin/ld: /opt/ros/humble/lib/librclcpp.so: undefined reference to `rcl_clock_time_started'
```

#### Additional information
Ever since PR #2210 has been merged, rclcpp now requires `rcl_clock_time_started` to be available in the rcl library, which was added [here](https://github.com/ros2/rcl/pull/1021). Currently there's no version check for rcl and now that the Patch 4 with this change is out it's possible to update rclcpp without updating rcl:
```
Package: ros-humble-rclcpp
Version: 16.0.5-2jammy.20230721.204813
Priority: optional
Section: misc
Maintainer: Ivan Paunovic <ivanpauno@ekumenlabs.com>
Installed-Size: 3342 kB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 3.3.1), libstdc++6 (>= 12), ros-humble-ament-index-cpp, ros-humble-builtin-interfaces, ros-humble-libstatistics-collector, ros-humble-rcl, ros-humble-rcl-interfaces, ros-humble-rcl-yaml-param-parser, ros-humble-rcpputils, ros-humble-rcutils, ros-humble-rmw, ros-humble-rosgraph-msgs, ros-humble-rosidl-runtime-cpp, ros-humble-rosidl-typesupport-c, ros-humble-rosidl-typesupport-cpp, ros-humble-statistics-msgs, ros-humble-tracetools, ros-humble-ros-workspace
```

I scoured the ros2 infrastructure stuff and can't find where dependency versions are defined or how they're even built [(using cpack or bloom)](https://github.com/ros-infrastructure/ros_buildfarm/blob/master/ros_buildfarm/scripts/release/build_binarydeb.py), so I hope it's the correct place for this issue. 

If someone tells me where this is defined, I can even open a PR for this fairly minor issue.

fujitatomoya, 2023-07-28:
I am not familiar with versioning process, but https://www.ros.org/reps/rep-0140.html#build-depend-multiple could be something?

CC: @ros2/team

sloretz, 2023-07-28:
`version_gte="VERSION"` attribte on the tag that depends on `rcl` should fix it - at least that's what I recall us doing for ROS 1

stan-guer, 2023-07-29:
@sloretz  you mean version 5.3.4? rcl has a different versioning: [cc9ea2810ff7c181ccdd43dc1f5c336f531fa8d8](https://github.com/ros2/rcl/commit/cc9ea2810ff7c181ccdd43dc1f5c336f531fa8d8) 

Wouldn't this fix also be applicable to rolling?

clalancette, 2023-07-31:
> Wouldn't this fix also be applicable to rolling?

No, I don't think we shouldn't apply these kinds of fixes to the `rolling` branch.  Rolling changes all of the time, and this metadata is going to get out-of-date really quickly.

billamiable, 2023-08-01:
We also experience build failure in ROS2 Humble complaining that

> /usr/bin/ld: /opt/ros/humble/lib/librclcpp.so: undefined reference to `rcl_clock_time_started'

How shall we proceed? Will it be fixed in the rclcpp or we can add some changes to our code to avoid this issue?