ros2/rclcpp #2253: Potential clock duplication in TimeSource

luca-della-vedova, 2023-07-31:
Hello! I was working on an rcl client library and was using `rclcpp` (and `rclpy`) as reference implementation, and when working on `TimeSource` I found a potential issue.

Specifically, when looking at the `TimeSource` implementation for both rclpy and rclcpp, I noticed that for each time source, its associated clocks are stored in a [vector](https://github.com/ros2/rclcpp/blob/22a954e1b00ea2ba810666baa92a3c66fe1a8654/rclcpp/src/rclcpp/time_source.cpp#L188).

Now apart from the premature-optimization-root-of-all-evil issue of having finds be linear in time [here](https://github.com/ros2/rclcpp/blob/22a954e1b00ea2ba810666baa92a3c66fe1a8654/rclcpp/src/rclcpp/time_source.cpp#L107), I believe the fact that clocks are naively appended in `attachClock` would allow duplicates in the [vector](https://github.com/ros2/rclcpp/blob/22a954e1b00ea2ba810666baa92a3c66fe1a8654/rclcpp/src/rclcpp/time_source.cpp#L98). Now the drawback is not the end of the world since it's just that the same clock would be updated more than once for each subscription callback but would it make sense to either:

* Do an `std::find` call in `attachClock` and print a warning + avoid appending if the clock already exists
* Make the attached clocks storage structure a `unordered_set`?

If any of these makes sense I'll be happy to open upstream PRs in both `rclpy` and `rclcpp` since they have the same implementation (rclpy is [here](https://github.com/ros2/rclpy/blob/913afa019b0d60e7255bd84247dbeab4735d96ea/rclpy/rclpy/time_source.py#L116))

alsora, 2023-07-31:
I'm not very familiar with `TimeSource`, but it seems to me that changing the `vector` into an `unordered_set` would be a good improvement.

clalancette, 2023-07-31:
I will say that from a performance POV, use of a `std::vector` probably isn't a problem.  I don't think there are common cases where we are attaching more than one `Clock` to a `TimeSource` (though if you have counter-examples, I'm happy to be proven wrong).

However, I do agree that storing a clock more than once in that vector seems unnecessary.  From that perspective, I'm also fine with switching that to be a `std::unordered_set`.

fujitatomoya, 2023-07-31:
> I do agree that storing a clock more than once in that vector seems unnecessary. From that perspective, I'm also fine with switching that to be a std::unordered_set.

So do I.

> I don't think there are common cases where we are attaching more than one Clock to a TimeSource

I dont quite figure out the use case for this either, any thoughts?

luca-della-vedova, 2023-08-01:
> > I don't think there are common cases where we are attaching more than one Clock to a TimeSource
> 
> I dont quite figure out the use case for this either, any thoughts?

I honestly don't know and I can't come up with any either, I was just reviewing the API and bumped into this corner case.
If having multiple clocks attached to a time source is a use case that isn't supported then another possibility could be for the time source to just store a single clock pointer in it and have the attach / detach APIs overwrite it.

clalancette, 2023-08-01:
> If having multiple clocks attached to a time source is a use case that isn't supported then another possibility could be for the time source to just store a single clock pointer in it and have the attach / detach APIs overwrite it.

While I sympathize with this idea, the use of a vector seems deliberate (and has been there in all incarnations of this code).  Maybe @tfoote can answer why `associated_clocks` is a vector, since he wrote the original code.

My suggestion is to switch this to a `std::unordered_set` for now.  It may be overkill (depending on the results of the conversation above), but it at least keeps the possibility to have multiple clocks attached to a TimeSource.

fujitatomoya, 2023-08-01:
@luca-della-vedova are you willing to work on PR? i am happy to do review.

PR #2257 (2023-08-03): Change TimeSource associated clocks storage to unordered_set
Fixes #2253 

As mentioned in the attached issue, using a `vector` makes it possible for the same clock to be stored multiple times in a `TimeSource`, this PR changes the storage to be `unordered_set` so duplicates are automatically avoided at insertion and removal is constant in time.

I thought a bit about adding a regression test but it wasn't so immediate to me it would be a good idea, the test would have been adding the same clock twice, removing it once then making sure it is not updated anymore. I can add it if you folks think [... cut]
closed 2023-08-03 by clalancette