ros2/rclcpp #2264: Parameters in YAML not loaded if not declared by declare_parameter()

KKSTB, 2023-08-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - iron
- DDS implementation:
  - NA
- Client library (if applicable):
  - rclcpp lifecycle_node parameter

#### Steps to reproduce issue
I am porting my custom node from humble to iron. There is an important parameter that must be declared in YAML. So my node will check whether the parameter exists in YAML using `bool get_parameter(const std::string & name, ParameterT & parameter)` and would not continue to init if not. However, the function fails even though the parameter is in YAML.

I further test the behaviors:
1. Declare extra dummy parameters in YAML --> these parameters are neither listed in `ros2 param list` nor in rqt dynamic reconfigure
2. Call `auto declare_parameter(const std::string & name, const ParameterT & default_value)` for the important parameter with some default value --> the parameter now listed in `ros2 param list` and in rqt dynamic reconfigure, with value in YAML
3. With step 2 but remove the parameter in YAML, the parameter now takes the default value (as expected)
#### Expected behavior
1. Parameters in `bool get_parameter()` should return true if they are in YAML, without needing the call to `declare_parameter()`
2. Extra parameters in YAML not declared by `declare_parameter()` should be listed in `ros2 param list`
#### Actual behavior
1. `bool get_parameter()` returns false even though the parameter is in YAML
2. Extra parameters in YAML not declared by `declare_parameter()` are not listed in `ros2 param list`
#### Additional information
I also tried the above test on `Universal_Robots_ROS2_Driver` main branch, launched by command `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=yyy.yyy.yyy.yyy use_mock_hardware:=true launch_rviz:=true initial_joint_controller:=joint_trajectory_controller`:
1. Declare extra dummy parameters in YAML --> these parameters are neither listed in `ros2 param list` nor in rqt dynamic reconfigure
2. Change parameters in `ur_controllers.yaml` --> the changed value takes effect

Whereas in ROS humble of `Universal_Robots_ROS2_Driver` humble branch, launched by command `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true initial_joint_controller:=joint_trajectory_controller` (note the difference between `use_mock_hardware` and `use_fake_hardware`):
1. Declare extra dummy parameters in YAML --> these parameters are listed in `ros2 param list` and in rqt dynamic reconfigure

iuhilnehc-ynos, 2023-08-04:
> I am porting my custom node from humble to iron. 

I guess you didn't use `options` like these ([1](https://github.com/ros2/demos/blob/fab4fc38d5ea589c019659d6a49da4ba9c20594b/demo_nodes_cpp/src/parameters/parameter_blackboard.cpp#L36-L37), [2](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/controller_manager.cpp#L129-L130)) for your custom node.

Can you reproduce this issue by https://github.com/ros2/demos/blob/rolling/demo_nodes_cpp/src/parameters/parameter_blackboard.cpp?

KKSTB, 2023-08-07:
I dont know where to configure node options in lifecycle node, especially for nodes inheriting `controller_interface::ControllerInterface` and controlled by controller manager in particular. Is there any example to follow? Thanks.

KKSTB, 2023-08-08:
Thanks for the test code. After looking deep, it seems it is caused by https://github.com/ros-controls/ros2_control/pull/757/commits/8d3fff9765ddfcd12b8fc317d62fa6f8aaf77331

But it seems the value is built with the iron release of controller manager (https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/controller_manager.cpp#L1156), so my controller cannot modify the node option by overriding it.

I will file an issue to `ros2_control`. Thanks.

closed 2023-08-08 by KKSTB