ros2/rclcpp #2273: Can't enable set_logger_levels service for LifecycleNode

bijoua29, 2023-08-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - latest rolling as of 2023-08-14
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - rclcpp lifecycle_node

#### Steps to reproduce issue

The enable_logger_service for NodeOptions doesn't seem to be enabled for LifecycleNode
```
node_options.enable_logger_service(true);
```
Passing node_options to creation of LifecycleNode doesn't enable the logger service

#### Expected behavior
The set_logger_levels service should be available when doing a 'ros2 node info' on the lifecycle node

#### Actual behavior
The set_logger_levels service is not seen when doing a 'ros2 node info' on the lifecycle node

#### Additional information

I notice creating a non-lifecyle node i.e. a standard node,  has the following code in the constructor

```
if (options.enable_logger_service()) {
    node_logging_->create_logger_services(node_services_);
}
```
However this doesn't exist in the constructor for the lifecycle node.

bijoua29, 2023-08-16:
Hmm. Ok. Is it as simple as adding 

```
if (options.enable_logger_service()) {
    node_logging_->create_logger_services(node_services_);
}
```

in the LifecycleNode constructor or am I missing something?

fujitatomoya, 2023-08-16:
implementation would be straight-forward to add the services, but test is also ideal. i think i can work on this.

bijoua29, 2023-08-16:
> implementation would be straight-forward to add the services, but test is also ideal. i think i can work on this.

Thanks @fujitatomoya

clalancette, 2023-10-03:
This was solved by #2277, so I'm going to close this out.

closed 2023-10-03 by clalancette