ros2/rclcpp #2278: EventsExecutor Service hangs when calling another service

russkel, 2023-08-18:
## Bug report

**Required Info:**

- Operating System:
  - Jammy x86_64
- Installation type:
  - Binaries
- Version or commit hash:
  - ros-iron-rclcpp/jammy,now 21.0.2-1jammy.20230713.215706 amd64 [installed,automatic]
- DDS implementation:
  - FastDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

I have a node that uses the EventsExecutor, that used to use a MultiThreadedExecutor (with the service and client in a re-entrant callbackgroup).

It has a Service that calls a service in another node. The client call deadlocks nodes with SingleThreadedExecutors and it seems to deadlock on the EventsExecutor as well. 

I know this kind of service calling a service arrangement isn't recommended and I do intend on rewriting it, but seeing as the EventsExecutor uses the RMW callbacks, why does this deadlock? Am I fundamentally misunderstanding how the executor is working?

#### Expected behavior
Client handles response correctly so the Service returns as expected.

#### Actual behavior
Client sends request, (external node handles request and returns a response) but times out "waiting" for response.

alsora, 2023-08-18:
RMW callbacks have no impact on this, they are simply a more efficient way to notify the executor that new work is available (replacing the waitset).

Calling a blocking service within a service (or within any other ROS 2 callback e,g, subscription) will deadlock as long as the two entities involved are handled by the same executor thread.

An executor has a queue of incoming events (this is exactly how events-executor is implemented, other executors do not have a queue but the behavior is equivalent).
A single threaded executor will sequentially take an element from the queue and process it (i.e. you receive a ROS 2 message, the executor calls the associated subscription callback).
While you are processing an element, any other incoming event will be blocked.

There are many solutions around this.
Notice how I said
> deadlock as long as the two entities involved are handled by the same executor thread
the easy solution is to have multiple executors or multiple threads.
this can be achieved in different ways:

1. use a multi-threaded executor
2. assign the two services to different callback groups and have them be handled by different executors

The other alternative is to not have blocking services within services.

russkel, 2023-08-18:
Thanks @alsora, my understanding was wrong and I did a bit of reading after I posted this.

> assign the two services to different callback groups and have them be handled by different executors

I believe I tried this thinking it would solve it and it didn't. I will try this again as perhaps my build environment was broken (tests were running a stale binary).

russkel, 2023-08-18:
I did re-attempt with service and client in separate callback groups with the EventsExecutor and it still deadlocks. I assume your advice was use two callback groups AND a multithreaded executor.

alsora, 2023-08-18:
You need to create two callback groups and manually add them to different executors

Executors have two APIs: `add_node` and `add_callback_group`.
When you call `add_node`, the executor will get access to all the entities in the node that are not in a callback group labelled as "do not automatically add".
With `add_callback_group` on the other hand you can manually add specific groups to an executor.

Having the entities in different groups is a requirement to be able to add them to different executors using the `add_callback_group` API.

russkel, 2023-08-18:
That makes sense now. Thanks for that explanation. I added another EventsExecutor and did what you said and it works. I had never seen the `add_callback_group` method before.

Thanks for your help and looking forward to the multi thread support in EventsExecutor.

closed 2023-08-18 by russkel

rty813, 2025-06-10:
> Note: it's possible to extend the events-executor to be multi-threaded, it's on our backlog but we haven't started working on it yet.

@alsora Where can I track this progress?