ros2/rclcpp #2286: Add descriptions to all node ROS interfaces

jonselling, 2023-08-25:
## Feature request

#### Feature description

Similar to the ParameterDescriptor, it would be good to create a Descriptor type that can be used for other ROS interfaces. 
These interfaces would include at minimum Publishers and Subscribers, but hopefully can support Services and Actions.

Primarily, this would just be associating a description string with the interface itself.
There might also be some additional meta-data we might deem useful such as expected frequency.

Why this change can be beneficial is we can work towards creating an auto documentation tool to document all ROS interfaces for a single Node.
With these added descriptions and the existing description interface for parameters and being able to get them through the describe_parameters interface, an auto documentation tool can be written to scrape all existing interfaces + parameters and a string description for everything. This would make node level documentation extremely easy to aggregate.

#### Implementation considerations

I was initially thinking we would make another message type in rcl_interfaces similar to the ParameterDescriptor, something like InterfaceDescriptor.

I was also thinking about making it a different service to call entirely and not conflicting with the "describe_parameters" interface.

fujitatomoya, 2023-08-28:
@jonselling thanks for posting feature request.

i am not sure what description to add and how the application take advantage of those... can you add a few more examples how this works for user application?

> Primarily, this would just be associating a description string with the interface itself.

Why this needs to be implemented as API? If that is for document generation, can we just add comments in the doc sec for the node source code? we can see the message dependencies and add doc sec for more explanation for each message types as node implementation details?

> There might also be some additional meta-data we might deem useful such as expected frequency.

expected frequency sometimes needs to be adjusted during runtime, i do not think this is statically set in the node. besides, i think this is irrelevant to data structure, but implementation detail. i would have the parameter named `current_publisher_rate`, so that user application can get and set the publication rate control?

jonselling, 2023-08-29:
So by "description to add" I am more explicitly saying that if we create a ROS interface I want to add a description to that interface in the source code so I as the developer can describe what that interface may expect.

> If that is for document generation, can we just add comments in the doc sec for the node source code

Documentation in this sense is ROS interface documentation. This is very similar to the "Node" section in the wiki.ros.org documentation you might see for some packages (at least for ROS 1). Depending on the complexity of the node you are writing, there may not be a clean mapping of source code in a single package will encompass all of the ROS interfaces your node creates.

An example of this would be if your node has dynamically loadable plugins. We will create publishers and subscribers that can be outside of the package, with no knowledge of the source code. Our system will have the plugins configured the same way every time, and we would like to document all of the ROS inputs and outputs.
This would also give the ability for whoever developed the plugin to add description strings to the interfaces they created.

> Why this needs to be implemented as API?

This would give people the flexibility to add that description when creating the interface, who would know best the intentions of why we are subscribing or publishing that data.

> expected frequency sometimes needs to be adjusted during runtime, i do not think this is statically set in the node.

This was more of an example of in addition to just a string description, we might want to encode an expected frequency. And think of it as more useful from the subscriber side (I expect to have this message continuously at this minimum rate). This might not have been the best example since diagnostics may be better suited for capturing topic frequency and if there is a possible error.

fujitatomoya, 2023-08-29:
> Our system will have the plugins configured the same way every time, and we would like to document all of the ROS inputs and outputs.

either plugins or not, we can see the ROS 2 node in the network.
how about using `ros2 node list`, `ros2 node info`, `ros2 interface show xxx` to get all interfaces? that is we can do with current ROS 2 distributions.

> This would also give the ability for whoever developed the plugin to add description strings to the interfaces they created.

let's say we add the description field to the message interface by default (or service to provide this description). i believe we cannot expect all developers and application implement the description field, that means we cannot still get what we need?

> And think of it as more useful from the subscriber side (I expect to have this message continuously at this minimum rate)

topic message type defines the data structure but configuration data. i would use parameter instead, to know and control the publication rate, that is more flexible for the application.

btw, if i am not mistaken, i would use the topic and service paths to add these `meaning`. (for example, `/robot/face/face_control/eyes/left` and `/robot/face/face_control/eyes/right` with the same message type `image`. again, this needs to be done in the application including remapping.)

> since diagnostics may be better suited for capturing topic frequency and if there is a possible error.

agree. i think diagnostics and deterministic needs to be based on the data what it is actually happening but expectation. there is QoS deadline that we could use for this case, see more details for https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Quality-of-Service-Settings.html#qos-policies

we can keep this open to get more feedback.

jonselling, 2023-08-31:
> how about using ros2 node list, ros2 node info, ros2 interface show xxx to get all interfaces?

Yes, this gets us like 90% of the way there for documentation. The last part of the ROS node interface documentation is usually adding a string description.

> i believe we cannot expect all developers and application implement the description field, that means we cannot still get what we need?

If they don't want to fill it in, they don't have to. I don't expect every developer to fill in the "ParameterDescriptor" for every parameter people declare.

> btw, if i am not mistaken, i would use the topic and service paths to add these meaning. 

Yes, I think that is good practice to make some more descriptive topic names.

> there is QoS deadline that we could use for this case,

This is a useful QoS field I've never played with, thanks for the info :+1: 

At this point my comments about topic frequency are causing confusion, so please just ignore them. It was more a pitch of what can maybe be added as an "InterfaceDescriptor" (that would be similar to a "ParameterDescriptor"). Just assume I am talking about adding a description string.

jonselling, 2023-08-31:
To add examples, I am thinking of making documentation very similar to these examples (but for ROS 2 nodes)

http://wiki.ros.org/image_view#image_view.2Fdiamondback.Nodes

http://wiki.ros.org/image_rotate?distro=noetic#image_rotate-1

http://wiki.ros.org/joint_state_publisher#Node_API

Adding a string description to all interfaces will have the documentation be pulled from the source directly when your node spins it up, so no need to fill it in manually.

clalancette, 2023-09-14:
We think that the idea of this is good, though there are no immediate plans to work on something like this.  There is some work in this direction in the NoDL stuff: https://github.com/ros2/design/pull/266

I'm going to move this to the ros2/ros2 repository, as it seems more appropriate there.  Feel free to continue the conversation.