ros2/rclcpp #2305: rclcpp compile error : rcl_node_type_description_service_handle_request’ was not declared in this scope

ericxiaoxin, 2023-09-12:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu22.04
- Installation type:
  - ros2 iron
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
git clone and git checkout iron, then colcon build, and error like: 
rclcpp/rclcpp/src/rclcpp/node_interfaces/node_type_descriptions.cpp: In destructor ‘rclcpp::node_interfaces::NodeTypeDescriptions::NodeTypeDescriptionsImpl::~NodeTypeDescriptionsImpl()’:
/home/ericma/ws/px_ros2_ws/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_type_descriptions.cpp:132:21: error: ‘rcl_node_type_description_service_fini’ was not declared in this scope
  132 |       RCL_RET_OK != rcl_node_type_description_service_fini(node_base_->get_rcl_node_handle()))
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

fujitatomoya, 2023-09-12:
@ericxiaoxin can you provide more information here, like complete procedure what you tried to build `rclcpp`.

clalancette, 2023-10-03:
No response in several weeks, so I'm going to close this out.  If you are still having issues, please open a question on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

closed 2023-10-03 by clalancette