ros2/rclcpp #2306: cannot link

ericxiaoxin, 2023-09-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu22.04
- Installation type:
  - ros2 from binaries, but i want to compile "rclcpp" from source
- Version or commit hash:
  - iron

#### Steps to reproduce issue
```
colcon build --allow-overriding rclcpp rclcpp_action rclcpp_components rclcpp_lifecycle
```

#### Actual behavior
```
Starting >>> rclcpp  
--- stderr: rclcpp                               
/usr/bin/ld: cannot find -lmimick: No such file or directory
/usr/bin/ld: cannot find -lmimick: No such file or directory
/usr/bin/ld: cannot find -lmimick: No such file or directory
/usr/bin/ld: cannot find -lmimick: No such file or directory
/usr/bin/ld: cannot find -lmimick: No such file or directory
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/rclcpp/CMakeFiles/test_expand_topic_or_service_name.dir/build.make:168: test/rclcpp/test_expand_topic_or_service_name] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1437: test/rclcpp/CMakeFiles/test_expand_topic_or_service_name.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/rclcpp/CMakeFiles/test_exceptions.dir/build.make:168: test/rclcpp/test_exceptions] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1185: test/rclcpp/CMakeFiles/test_exceptions.dir/all] Error 2
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/rclcpp/CMakeFiles/test_loaned_message.dir/build.make:192: test/rclcpp/test_loaned_message] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1632: test/rclcpp/CMakeFiles/test_loaned_message.dir/all] Error 2
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/rclcpp/CMakeFiles/test_client.dir/build.make:192: test/rclcpp/test_client] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1325: test/rclcpp/CMakeFiles/test_client.dir/all] Error 2
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/rclcpp/CMakeFiles/test_node.dir/build.make:192: test/rclcpp/test_node] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1716: test/rclcpp/CMakeFiles/test_node.dir/all] Error 2
/usr/bin/ld: cannot find -lmimick: No such file or directory
/usr/bin/ld: cannot find -lmimick: No such file or directory
/usr/bin/ld: cannot find -lmimick: No such file or directory
/usr/bin/ld: cannot find -lmimick: No such file or directory
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/rclcpp/CMakeFiles/test_node_interfaces__node_timers.dir/build.make:168: test/rclcpp/test_node_interfaces__node_timers] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1940: test/rclcpp/CMakeFiles/test_node_interfaces__node_timers.dir/all] Error 2
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/rclcpp/CMakeFiles/test_node_interfaces__node_interfaces.dir/build.make:168: test/rclcpp/test_node_interfaces__node_interfaces] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1856: test/rclcpp/CMakeFiles/test_node_interfaces__node_interfaces.dir/all] Error 2
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/rclcpp/CMakeFiles/test_node_interfaces__node_parameters.dir/build.make:168: test/rclcpp/test_node_interfaces__node_parameters] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1884: test/rclcpp/CMakeFiles/test_node_interfaces__node_parameters.dir/all] Error 2
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/rclcpp/CMakeFiles/test_node_interfaces__node_services.dir/build.make:168: test/rclcpp/test_node_interfaces__node_services] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1912: test/rclcpp/CMakeFiles/test_node_interfaces__node_services.dir/all] Error 2
```

fujitatomoya, 2023-09-12:
i think that because you are trying to build `rclcpp` without `rclcpp` test dependent packages.

can you try with,

```bash
colcon build --packages-up-to rclcpp rclcpp_action rclcpp_components rclcpp_lifecycle
```

clalancette, 2023-09-12:
> i think that because you are trying to build `rclcpp` without `rclcpp` test dependent packages.

Theoretically, this should work.  That is, it is a supported use case to overlay just `rclcpp` on top of binary packages, and successfully compile it.  The following should work:

```
$ . /opt/ros/iron/setup.bash
$ mkdir -p rclcpp_ws/src
$ cd rclcpp_ws
$ git clone -b iron https://github.com/ros2/rclcpp src/rclcpp
$ colcon build --allow-overriding rclcpp rclcpp_action rclcpp_components rclcpp_lifecycle
```

And indeed, it does work for me, on Rolling, Iron, and Humble, but with a caveat.  When I first tried this out, I had already done `source /opt/ros/iron/setup.bash`, but then I installed `ros-iron-mimick-vendor` as part of installing dependencies.  When I did *that*, I got the same error as @ericxiaoxin .  If I sourced /opt/ros/iron/setup.bash again, though, it started working.

So I would say that this is a bug, but probably not an urgent one because it does work by resourcing /opt/ros/iron/setup.bash