ros2/rclcpp #2314: Creating a Subscriber with TopicStatistics Applies Custom QoS to /statistics Publisher

JWhitleyWork, 2023-09-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - Humble, Iron
- DDS implementation:
  - Any
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
```
rclcpp::SubscriptionOptions opts{};
opts.topic_stats_options.state = rclcpp::TopicStatisticsState::Enable;

rclcpp::QoS bad_qos{};

this->creare_subscription<DummyMsg>("/some_topic", bad_qos.history(rclcpp::HistoryPolicy::Unknown), cb_func, opts);
```

#### Expected behavior
The `/statistics` publisher has a stable, well-defined QoS profile.

#### Actual behavior
The QoS profile passed to `create_subscription` is applied to both the subscription and the `/statistics` publisher. Code reference in [humble](https://github.com/ros2/rclcpp/blob/humble/rclcpp/include/rclcpp/create_subscription.hpp#L92) and [iron](https://github.com/ros2/rclcpp/blob/iron/rclcpp/include/rclcpp/create_subscription.hpp#L92).

fujitatomoya, 2023-09-24:
@JWhitleyWork that makes sense to me. adding `QoS` to `TopicStatisticsOptions` struct as `statistics` publisher QoS configuration.

CursedRock17, 2023-12-11:
Since that PR was merged this issue should be complete right?

clalancette, 2023-12-20:
Yeah, I'm going to close it.  Thanks!

closed 2023-12-20 by clalancette