ros2/rclcpp #2317: Consider renaming `pre_set`, `on_set` and `post_set_parameters_callback`.

jrutgeer, 2023-09-27:
PR https://github.com/ros2/rclcpp/pull/1947 added suport for `pre_set` and `post_set` parameter callbacks.

I think the naming of these callbacks is rather ill-chosen:

- The name `on_set_parameters_callback` suggests that this is when the parameters _are set_, or _should be set_, while in fact the internal state **should not** be altered,
- the `post_set` callback is not entirely clear to me:
  1. According to the [documentation](https://docs.ros.org/en/rolling/Concepts/Basic/About-Parameters.html#parameter-callbacks):
     -  "The main purpose of this callback is to give the user the ability to react to changes from parameters that have successfully been accepted."
     - So this suggests that the node parameters have been updated _before_ this call,
     - In which case it would indeed be a `post_set` callback,
   2. But in the [example code](https://github.com/ros2/demos/blob/36a4c0bd6ae0531ad3096806c41efd7062a1cb31/demo_nodes_cpp/src/parameters/set_parameters_callback.cpp#L86-L99), the `post_set` callback is used to update the node parameters.
      - So this seems contradictory to (i.), and it is rather a `process_new_parameters_callback` than a `post_set_parameters_callback`.

I assume this is not of highest priority, but still: if at some point other changes are made to the parameter functionality, I propose to consider renaming the callbacks, e.g.:

- `pre_set_parameters_callback` --> `modify_upcoming_parameters_callback`
- `on_set_parameters_callback` --> `approve_upcoming_parameters_callback`
- `post_set_parameters_callback` --> `react_to_updated_parameters_callback` or `process_new_parameters_callback` (depending on whether (i.) or (ii.) is the actual intended behavior).

Imo. such naming would better correspond to the functional description in the  [documentation](https://docs.ros.org/en/rolling/Concepts/Basic/About-Parameters.html#parameter-callbacks) and avoid confusion.

clalancette, 2023-09-27:
> * The name `on_set_parameters_callback` suggests that this is when the parameters _are set_, or _should be set_, while in fact the internal state **should not** be altered,

While I don't disagree, we've already renamed this callback 3 times.  And since this is the oldest of the callbacks, it has the most users.  So I'm not super inclined to change this one yet again.

> * But in the [example code](https://github.com/ros2/demos/blob/36a4c0bd6ae0531ad3096806c41efd7062a1cb31/demo_nodes_cpp/src/parameters/set_parameters_callback.cpp#L86-L99), the `post_set` callback is used to update the node parameters.

No, that example code does not change the node parameters.  It changes internal class state, which is a different thing.  That is, there is a (private) database of node parameters that rclcpp tracks, and by the time that `post_set` is called, that database has been updated.  The purpose of the `post_set` is to give the user the chance to make internal changes based on that, which is what the example shows.  Maybe we should update the name of the member fields in the example code, and/or update the documentation?

jrutgeer, 2023-09-28:
> So I'm not super inclined to change

That's why I used the wording "I propose to _consider_".   ;-)

> **this one** yet again.

I would rename all, or none. 
Or maybe add the new names as an alias and also keep the current ones?
Anyway if it is deemed not important enough than that's fine as well.

> No, that example code does not change the node parameters. It changes internal class state, which is a different thing.

Ok, I totally missed that the `internal_tracked_class_parameter_X_` variables are of type `double`.
I assumed that `this->declare_parameter()` would return a `rclcpp::Parameter`, similar to `this->get_parameter()`.
So I was expecting rather:

```
internal_tracked_class_parameter_1_ = this->declare_parameter("param1", 0.0).as_double();
[...]
internal_tracked_class_parameter_1_ = this->get_parameter("param1").as_double();
```

jrutgeer, 2023-09-28:
@clalancette Can you check this proposal https://github.com/ros2/demos/pull/651? See also the open questions in the comment.

closed 2023-11-08 by clalancette