ros2/rclcpp #2328: Adding rmw_request_id_t allocator

CursedRock17, 2023-10-06:
## Feature request
Should `rmw_request_id_t`'s allocator be used in the `create_request_header` method
#### Feature description

Added in [this commit](https://github.com/ros2/rclcpp/commit/209a31c8156d9ac312acfe1e13ef713ca02909df), there is a [TODO](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/include/rclcpp/client.hpp#L541) within the method `create_request_header` that wants to utilize [`rmw_request_id_t`](https://github.com/ros2/rmw/blob/rolling/rmw/include/rmw/types.h#L358-#L365) with a custom allocator of some sort.

#### Implementation considerations
This TODO is roughly 8.5 years old, a lot has changed since then especially with rmw, so if this isn't necessary I can just open a PR that removes the TODO. Otherwise, since `rmw_request_id_t` doesn't have an allocator inside of the class, should this method be using some from of allocator in `rclc` since it is a C type, in order to efficiently create new request headers?

fujitatomoya, 2023-10-27:
@CursedRock17 i am not sure what needs to be done here.

@wjwwood can you share your idea or example?

clalancette, 2023-10-31:
I'm not 100% sure, but my guess is that the idea is to honor the custom `rcutils` allocator if the user has set it.  So this is probably still valid?  Unclear to me.  I guess we can wait a little longer for @wjwwood to respond.